NanoVNA-Saver/nanovna-saver #293: 0.3.7Pre1 Dev Branch - 6c4fbe6c8dfdfc8deb8db8315d4620440043cfa7

RandMental, 2020-07-29:
**Describe the bug**
When Segment is more than 1 an index error occurs

[image]

[Debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4993232/Debug.txt)

zarath, 2020-07-29:
Yes I know. Can you please checkout v0.3.7-pre06

RandMental, 2020-07-29:
Apologies, not sure why Pycharm Git did not pickup the latest version.

RandMental, 2020-07-29:
NanoVNASaver 0.3.7-pre07  crashes and closes after connecting to H4 – 0.9.3.4 Firmware

 

 

The C:\Users\deond\PycharmProjects\untitled\venv\Scripts\python.exe -m NanoVNASaver.__main__ -D Debug.txt

 

NanoVNASaver 0.3.7-pre07

 

Copyright (C) 2019, 2020 Rune B. Broberg

Copyright (C) 2020 NanoVNA-Saver Authors

 

This program comes with ABSOLUTELY NO WARRANTY

This program is licensed under the GNU General Public License version 3

 

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

 

Settings: C:/Users/deond/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini

 

Process finished with exit code -1073740791 (0xC0000409)

 

2020-07-29 09:46:01,691 - NanoVNASaver - INFO - Startup...
2020-07-29 09:46:01,765 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 09:46:01,765 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-29 09:46:01,765 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 09:46:01,765 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-07-29 09:46:01,766 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-29 09:46:02,095 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-29 09:46:02,321 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.6"
2020-07-29 09:46:02,321 - NanoVNASaver.Windows.About - INFO - This is 0.3.7-pre07
2020-07-29 09:46:02,326 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-07-29 09:46:02,374 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-29 09:46:02,569 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-29 09:46:02,572 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-29 09:46:05,326 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-29 09:46:05,431 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-29 09:46:05,586 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-29 09:46:05,710 - NanoVNASaver.Hardware.Hardware - DEBUG - Needed retries: 0
2020-07-29 09:46:05,710 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-29 09:46:05,711 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-29 09:46:05,816 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 09:46:05,867 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 09:46:05,867 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-29 09:46:05,867 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-29 09:46:05,867 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-29 09:46:05,970 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 09:46:06,025 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 09:46:06,025 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['Commands:', 'version', 'reset', 'freq', 'offset', 'bandwidth', 'time', 'dac', 'saveconfig', 'clearconfig', 'data', 'frequencies', 'port', 'stat', 'gain', 'power', 'sample', 'scan', 'sweep', 'test', 'touchcal', 'touchtest', 'pause', 'resume', 'cal', 'save', 'recall', 'trace', 'marker', 'edelay', 'capture', 'vbat', 'vbat_offset', 'transform', 'threshold', 'help', 'info', 'color', 'i2c', 'lcd']
2020-07-29 09:46:06,025 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth)
2020-07-29 09:46:06,130 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 09:46:06,182 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
[... 65 lines cut ...]
2020-07-29 09:46:07,274 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 46
2020-07-29 09:46:07,988 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 5
2020-07-29 09:46:07,988 - NanoVNASaver.Hardware.NanoVNA - DEBUG - readFrequencies: scan
2020-07-29 09:46:07,988 - NanoVNASaver.Hardware.VNA - DEBUG - VNA reading frequencies
2020-07-29 09:46:07,988 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(frequencies)
2020-07-29 09:46:08,116 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 46
2020-07-29 09:46:08,242 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 09:46:08,242 - NanoVNASaver.Hardware.VNA - DEBUG - VNA done reading frequencies (201 values)
2020-07-29 09:46:08,242 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-07-29 09:46:08,242 - NanoVNASaver.Hardware.VNA - DEBUG - VNA reading data 0
2020-07-29 09:46:08,242 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(data 0)
2020-07-29 09:46:08,370 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 46
2020-07-29 09:46:08,527 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 09:46:08,527 - NanoVNASaver.Hardware.VNA - DEBUG - VNA done reading data 0 (201 values)
2020-07-29 09:46:08,527 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-29 09:46:08,527 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
2020-07-29 09:46:08,528 - NanoVNASaver.Hardware.VNA - DEBUG - VNA reading data 1
2020-07-29 09:46:08,528 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(data 1)
2020-07-29 09:46:08,654 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 46
2020-07-29 09:46:08,798 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 09:46:08,798 - NanoVNASaver.Hardware.VNA - DEBUG - VNA done reading data 1 (201 values)
2020-07-29 09:46:08,798 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-29 09:46:08,799 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-07-29 09:46:08,801 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-07-29 09:46:08,802 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 201, Offset: 0

RandMental, 2020-07-29:
NanoVNASaver 0.3.7-pre07  crashes and closes after connecting to H4 – 0.9.3.4 Firmware

 

 

The C:\Users\deond\PycharmProjects\untitled\venv\Scripts\python.exe -m NanoVNASaver.__main__ -D Debug.txt

 

NanoVNASaver 0.3.7-pre07

 

Copyright (C) 2019, 2020 Rune B. Broberg

Copyright (C) 2020 NanoVNA-Saver Authors

 

This program comes with ABSOLUTELY NO WARRANTY

This program is licensed under the GNU General Public License version 3

 

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

 

Settings: C:/Users/deond/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini

 

Process finished with exit code -1073740791 (0xC0000409)

zarath, 2020-07-29:
deleted dubug in comment text, as crash was not visible...
Can you please attach debug.txt as file and not as pasted text in future?

RandMental, 2020-07-29:
Interesting, I did add it as a file

 

2020-07-29 09:58:00,009 - NanoVNASaver - INFO - Startup...
2020-07-29 09:58:00,085 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 09:58:00,085 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-29 09:58:00,085 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 09:58:00,086 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-07-29 09:58:00,086 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-29 09:58:00,431 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-29 09:58:00,981 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.6"
2020-07-29 09:58:00,982 - NanoVNASaver.Windows.About - INFO - This is 0.3.7-pre07
2020-07-29 09:58:00,987 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-07-29 09:58:01,037 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-29 09:58:01,242 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-29 09:58:01,246 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-29 10:04:58,157 - NanoVNASaver - INFO - Startup...
2020-07-29 10:04:58,222 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 10:04:58,223 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-29 10:04:58,223 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-07-29 10:04:58,223 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-07-29 10:04:58,223 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-07-29 10:04:58,542 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-07-29 10:04:59,045 - NanoVNASaver.Windows.About - INFO - Latest version is 0.3.6"
2020-07-29 10:04:59,046 - NanoVNASaver.Windows.About - INFO - This is 0.3.7-pre07
2020-07-29 10:04:59,050 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-07-29 10:04:59,096 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-07-29 10:04:59,301 - NanoVNASaver.Hardware.Hardware - DEBUG - Found NanoVNA USB:(0483:5740) on port COM25
2020-07-29 10:04:59,304 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-07-29 10:05:05,985 - NanoVNASaver.NanoVNASaver - INFO - Connection COM25 (NanoVNA)
2020-07-29 10:05:06,090 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-07-29 10:05:06,244 - NanoVNASaver.Hardware.Hardware - INFO - Finding firmware variant...
2020-07-29 10:05:06,352 - NanoVNASaver.Hardware.Hardware - DEBUG - Needed retries: 0
2020-07-29 10:05:06,352 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-H4
2020-07-29 10:05:06,352 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(version)
2020-07-29 10:05:06,455 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 10:05:06,506 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:06,506 - NanoVNASaver.Hardware.VNA - DEBUG - result:
['0.4.5-23-g587ad32ab']
2020-07-29 10:05:06,506 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-29 10:05:06,506 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-29 10:05:06,610 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 10:05:06,663 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:06,663 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-29 10:05:06,663 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth)
2020-07-29 10:05:06,766 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 10:05:06,818 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:06,818 - NanoVNASaver.Hardware.NanoVNA - DEBUG - Using new scan command.
2020-07-29 10:05:06,818 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(info)
2020-07-29 10:05:06,922 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 26
2020-07-29 10:05:06,977 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:06,977 - NanoVNASaver.Hardware.VNA - DEBUG - result:
2020-07-29 10:05:06,977 - NanoVNASaver.Hardware.NanoVNA_H4 - INFO - VNA has 201 datapoints capability
2020-07-29 10:05:06,977 - NanoVNASaver.Hardware.VNA - DEBUG - Features: {'Scan command', 'Customizable data points', 'Screenshots', 'Bandwidth'}
2020-07-29 10:05:06,977 - NanoVNASaver.Hardware.VNA - DEBUG - get bandwidths
2020-07-29 10:05:06,978 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(bandwidth 0)
2020-07-29 10:05:07,081 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 47
2020-07-29 10:05:07,132 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:07,132 - NanoVNASaver.Hardware.NanoVNA_H4 - DEBUG - read_features
2020-07-29 10:05:07,132 - NanoVNASaver.Hardware.VNA - DEBUG - exec_command(help)
2020-07-29 10:05:07,236 - NanoVNASaver.Hardware.VNA - DEBUG - Max retries: 46
2020-07-29 10:05:07,288 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:07,288 - NanoVNASaver.Hardware.VNA - DEBUG - result:
[... cut ...]
2020-07-29 10:05:28,793 - NanoVNASaver.Hardware.VNA - DEBUG - Needed retries: 0
2020-07-29 10:05:28,793 - NanoVNASaver.Hardware.VNA - DEBUG - VNA done reading data 1 (201 values)
2020-07-29 10:05:28,793 - NanoVNASaver.SweepWorker - DEBUG - Read 201 values
2020-07-29 10:05:28,793 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-07-29 10:05:28,796 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-07-29 10:05:28,797 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 201, Offset: 0

igor-m, 2020-07-29:
I get the List.. error now at initial sweep with 1 segment in 037-pre7 as well, updated the original List.. bug..

zarath, 2020-07-31:
related to #301

closed 2020-08-13 by zarath