NanoVNA-Saver/nanovna-saver #296: Nanovna-saver 0.3.7 rc08 crashes after  "Connect to device"

PA3CCE, 2020-07-29:
RC08 crashes as well.

[debug-rco8.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4994657/debug-rco8.txt)

zarath, 2020-07-29:
Don't see the crash. But funny frequencies on your device
2020-07-29 13:45:08,855 - NanoVNASaver.NanoVNASaver - INFO - Read starting frequency 800 and end frequency 2800

PA3CCE, 2020-07-29:
Well I can't help, didi it again after deleting Nanovnasaver depository on C:, new debug file , different ?

[dbug2_rc08.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4994700/dbug2_rc08.txt)

zarath, 2020-07-29:
Nothing wrong in debug file.. did it show anything on the command prompt?

PA3CCE, 2020-07-29:
No not trhat I know off , will check again.

PA3CCE, 2020-07-29:
Started comaand window again navigated to nanovna-saver-Development executed command nanovnasaver without debug.txt NanoVNA-Saver started up, pressed "Connect to device" and program **did not abort !**
Start stop settings 800/2800.
Maybe because I had hhave a run  with deleted NanoVNA repository before ???

PA3CCE, 2020-07-29:
Have set for sweep range 3M- 2G loaded caliubration from rc6 set for 20 steps, nice results

PA3CCE, 2020-07-29:
Even could sweep with averaging 9/4, result a bit less noisy, But after 100% sweep pressing "Display Setup" does not responjd.
That's not frendly when you want to see different displays.

PA3CCE, 2020-07-29:
Program is aborting when started with -D dbugXY.txt !!!!
Do I have to make a nieuw report for the Disp[lay setup problem ?

zarath, 2020-07-29:
yes...

closed 2020-07-29 by zarath

PA3CCE, 2020-07-29:
OK, but now, when I set for 3M/2G 30 steps 9/4 average, 10Hz bandwith the normal sweep works (very slow), but when I try a calibration with these settings it aborts.