NanoVNA-Saver/nanovna-saver #297: NanoVNA-Saver 0.3.7 rc08 conditional abort

PA3CCE, 2020-07-29:
**Describe the bug**
I downloaded and installed the very latest release from the development (updated 3hours ago). 
When I set for start/stop 3M/2G 20 Segments, Averged sweep set tp 9/4, Bandwith 10Hz, and try to do a calibration the program aborts.

[debug-rc08-new.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4995492/debug-rc08-new.txt)

PA3CCE, 2020-07-29:
end of the command window shows ;
 
Traceback (most recent call last) :
fale d:\programming\python 3.8.3\lib\site-packages\NanoVNASaver\Windows\CalibrationSettings.py",line 645, in automaticCalibration
    if self.app.windows["sweep-settings"].continuous._radiobutton.isChecked():
AtributeError: 'Sweepsettingswindow' object has no attribute 'continuous-sweep-radiobutton'

PA3CCE, 2020-07-29:
Found rc09, now I'm testing it.

PA3CCE, 2020-07-29:
well starting and running OK, but same problem with trying to calibrate
[debug-rc09.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4995766/debug-rc09.txt)

zarath, 2020-07-29:
it seems you find every single path in the code ... ;-)

PA3CCE, 2020-07-29:
Well, then finally we have a nice product :-)

zarath, 2020-07-29:
and what's about pre10?

PA3CCE, 2020-07-29:
OK, in the mean time I had dinner.
pre 10 calibration is running now, **did not abort,** very very slow as expected, I have set for 8 markers again and all are filled now, **vy nice,** but first 3 markers still @ 3- , 52.xxx and 102.xxx MHz and the remaining 5 markers @ 45.xxx, -59.xxx, 73.xxx, 88.xxx and 102.xxx MH, while frequency span is set for 3MHz to 2Ghz.. So that should  be better.
Will be back when calibration has finished.

PA3CCE, 2020-07-29:
Now the calibration run for the short has finished, took extremly long with these settings, I would'nt advise that, but I want to see what accuracy (noiselevel) is possible when you use all your options and have the time  :-). Anyhow the calibration problem seems to have been solved and I think we could close this item. Will do some more testing and when I find something I will let you know, thanks vy much for your effords.

zarath, 2020-07-29:
Ah, now I understand why are you getting all the edge cases.
You want to try what's the limits of your device ....
Stability seems to settle.... mostly cosmetic bugs now.... time to go to a rc version

closed 2020-07-29 by zarath