NanoVNA-Saver/nanovna-saver #298: NanoVNA-Saver RC10 poor calibration result

PA3CCE, 2020-07-29:
Calibration as described before (took hours), the result was poor ( NanoVNASaver.windows.sweepsettings - WARNING - illegal averaging values, set default).

[debug-rc09.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/4997983/debug-rc09.txt)

zarath, 2020-07-30:
2020-07-30 00:57:34,207 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-07-30 00:57:34,207 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-07-30 00:57:34,207 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3000000, 2000000000, 201, 20, Properties('', SweepMode.AVERAGE, (9, 4), False))
2020-07-30 00:57:34,207 - NanoVNASaver.SweepWorker - INFO - 9 averages
2020-07-30 00:57:34,208 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 9
2020-07-30 00:58:07,569 - NanoVNASaver.SweepWorker - DEBUG - Stopping averaging as signalled

Something has called the stop signal during the averaging sweep....
That causes the error ....
2020-07-30 00:58:07,569 - NanoVNASaver.SweepWorker - DEBUG - Stopping averaging as signalled
2020-07-30 00:58:07,575 - NanoVNASaver.SweepWorker - DEBUG - Truncating 1 values by 4
2020-07-30 00:58:07,575 - NanoVNASaver.SweepWorker - DEBUG - Truncating from 1 values to -3
2020-07-30 00:58:07,575 - NanoVNASaver.SweepWorker - INFO - Not doing illegal truncate
2020-07-30 00:58:07,575 - NanoVNASaver.SweepWorker - DEBUG - Truncating from 1 values to -3
2020-07-30 00:58:07,575 - NanoVNASaver.SweepWorker - INFO - Not doing illegal truncate

PA3CCE, 2020-07-30:
Not me as far as I know. At that time I had to wake up the system several times to see how far the calibration was, that's all.
Will not repeat it, takes too long, maybe with 3/0 averaging if doable.

zarath, 2020-07-30:
> Not me as far as I know. At that time I had to wake up the system several times to see how far the calibration was, that's all.
> Will not repeat it, takes too long, maybe with 3/0 averaging if doable.

Perhaps by accident during wake-up.. I see no possible way to emit that stop signal than hitting the stop button...
Anyway, now just that dataset / segment will be discarded.

For precission test, I suggest trying less segments and high averages....

PA3CCE, 2020-07-30:
OK, I think we can close this issue now.
I did a successful calibration today with rc1 and the same parameters except the averaging, now set to 3/0.. Took long but not as extreme as yesterday..., In the mean time I could  life watch the launching of the new Mars rover on the screen  :-))....Calibration and measuring with averaging ruduces the noiselevel significantly espescial at frequencies above 1500MHz..... I attach here the result of a calibration with a 10dB attenuator on CH1, looks nice up to 1.5 G..... Don't know if the irregularity  at 1.5GHz is caused by the attenuator, but have a strong feeling that the harmonics switching in the firmware adds to it. 
[image]

I also found that in rc1 the markerfrequencies are now set for all markers that  I have set (8), but they are not calculated confirm the frequency span, you can see that in the screenshot.../.. Do I have to make a new issue for that ?.... Regards.

closed 2020-07-30 by zarath