NanoVNA-Saver/nanovna-saver #307: NanoVNA-Saver 0.3.7 rc05 results  minor as requested (minor problems)

PA3CCE, 2020-08-01:
**Describe the bug**
I performed 2 sweeps 10MHz-1.6 GHz different settings and I expexted better results iin the second sweep, but this is not (non solved earlier reported  problems),, and one error meassage (minor problem).

**To Reproduce**

Steps to reproduce the behavior:

1.Start NanoVNA-Saver from Command Window... (Sweep settings are set to last used.)
2. Click press "Connect to device"  : Error Window pops up :
   Error during sweep .
   Stpped
3. Clicked OK
   Start/Stop still at lst set, 
   no markers filled in.
4. Pressed " Sweep".  (Sweep starts for 1 segmkent.)
   Markers are not filled in., and sweep finishes.
5. "Pressed Display setup" and set for 8 Markers. Marker fields are filled with frequencies devided over last used span (10M/1.8G).
6. Pressed "Manage" and set for 201 points and 100Hz bandwidth. No idea what really is set).
7.Set " Segments to 10).
8. Load calibration.  ( From last used release).
6. Start sweep  ( Sweep is ok, but Markers are reset to equally devided over 10 to !00MHz).
7. Pressed " Set current as a reference". (Colors are White and Rose)
7.Pressed "manage **( this takes extreemly long time and had to do several times before it popped up)**
8 Set for 10 Hz Bandwdth.
9. Pressed "Sweep Settings" and set for 9/4 averaging.
10. Press "Sweep",  sweep starts and takes long (as expected).
11. Sweep finishes correctly.
12.  Both sweeps are OKl, I expexted that the 2nd sweep was s;lightly better, **but on the contrary** ( I suppose the second          sweep was done at 4 kHz bandwidth, first on at ).

**Conclusion :** This release isrunning fine except for the marker settings (known problem), the B and the Bandwidth settings (known as well and the minor problem of the Error pop up.
This all is done to my best sikils, hope it is of use.  Screenshot is attached.

**13. Found after writing the previous : I have the same problem opening the "Manage" Window, takes long and aborts afterwards.**

[image]

zarath, 2020-08-01:
Regarding bandwidth, I think we have to do a feature request to DiSlord.
There is currently now way to read out, which bandwidth setting value results in
which bandwidth. It depends on the actual firmware used.
That 4k needs different values than the 2k version :-(
I have currently no clue (and no -H4)  how to resolve this issue.

But thanks for the report

PA3CCE, 2020-08-01:
Yes I do understand that , but I am using the vy latest firmware for the H4 users, I I'm counting on it that NanoVNA-H4 ussers with Dislord's firmware will load the very latest one. That manage problem is a strange one. I did not attach the debug file, have probl;ems with my mouse, acting wierd. Will try again.

PA3CCE, 2020-08-01:
Foud it, here is the Debug file 
[debug-rc05.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5010714/debug-rc05.txt)

Mike4U, 2020-08-03:
Looks like the button update is not happening same as on Mac.  The marker button should say 'Hide data' when data is shown and the serial connect button should say 'Disconnect'

zarath, 2020-08-03:
@Mike4U is moving of the border between marker values and charts possible?

On linux I can move the border as far to the left, that the markers disappear.
When I drag that border right they will be shown again. So I wonder if that show data/hide data is still of future use....

zarath, 2020-08-13:
Some troubles with button on mac should be solved with v0.3.7

so I close this issue and looking forward to get new ones with the new release ;-)

closed 2020-08-13 by zarath