NanoVNA-Saver/nanovna-saver #309: AVNA errors on connect

Mike4U, 2020-08-03:
**Describe the bug**
on connecting get this in terminal:

> OSError: too many retries
2020-08-03 10:35:25,821 - NanoVNASaver.SweepWorker - ERROR - too many retries
Traceback (most recent call last):
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/SweepWorker.py", line 80, in run
    self._run()
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/SweepWorker.py", line 122, in _run
    start, stop, averages)
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/SweepWorker.py", line 236, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/SweepWorker.py", line 260, in readSegment
    self.app.vna.setSweep(start, stop)
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/Hardware/AVNA.py", line 42, in setSweep
    list(self.exec_command(f"sweep {start} {stop} {self.datapoints}"))
  File "/Users/michaelrunyan/Documents/GitHub/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 100, in exec_command
    raise IOError("too many retries")
OSError: too many retries
> 

This used to work in an older version.
**Desktop (please complete the following information):**
 - OS: Mac 10.15.6 Catalina
 - Python version: 3.7.3
 - NanoVNA-Saver version: 0.3.7-rc06

How many people use this ?  I'll try to fix when I get a chance.

zarath, 2020-08-04:
Ok, if it worked in 0.3.4 there is a chance to fix it...
Can you create debug file and attach it here? 
`python nanovna-saver.py -D debug.txt`

Mike4U, 2020-08-04:
I'm not sure the 0.3.4 is correct.  I'm trying to verify.  It did work on 0.2.3 for sure, have a pic.

Here's the debug from 0.3.7-rc06
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5022941/debug.txt)

zarath, 2020-08-04:
Do you know how fast the AVNA do a 101 point scan?
You can try to increase 
WAIT = 0.05

in Hardware/VNA.py

atm. it expects to get the 101 points in about 2 seconds.

And I see, that I have to restructure the code segment a little and perhaps move the retry calculation function to vna class method. ...

Mike4U, 2020-08-04:
AVNA is super slow and typically does 13 points from 10 to 40kHz in about 8 seconds.  Higher the frequency, faster the scan.  10Hz and 40kHz are the minimum and maximum.
I'll try the WAIT.

zarath, 2020-08-04:
OK. That 40times slower use 2.00 as wait... But as I told. It needs some restructuring for a final fix

Mike4U, 2020-08-04:
Set WAIT to 2.0
Result: nothing, at least it didn't crash.  Set for 11 points and tried again, same, nothing happens.
Here's the debug:
[AVNA_debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5023604/AVNA_debug.txt)

gotta go

Mike4U, 2020-08-04:
Tried with WAIT 4.0 still errors after longer wait.
[image]

zarath, 2020-08-05:
OK. Strange.. it seems that no data is arriving after the sweep command.. 
don't know if there is some more delay between issuing the sweep command and reading first data needed. 
I'll take a look after a small weekend vacation... So no time for this before middle of next week.

Mike4U, 2020-08-05:
I am looking at the AVNA code and finding errors.  Got past the sweep issue, now stuck at frequency.  I'll try to get help from AVNA author.  Not sure its worth you looking at right now.

closed 2022-01-04 by zarath