NanoVNA-Saver/nanovna-saver #314: Setup when starting the nanovna-saver

PY3IG, 2020-08-07:
- When we only have two active graphics, it is necessary to scale the window size when starting the nano-saver. (Figures 1 and 2)

- Is it possible to hide the markers when starting? They are present in the graph immediately when connecting the nanovna to the nanovna-saver. (Figure 3). I want to use markers, but I would like to see when selected.

These two cases are not manifested in nanovna-Saver 0.3.6

Thank you for your work
73!
Igor Jobim
[image]
[image]
[image]

G4GCL, 2020-08-14:
These problems are also on the final version of V0.3.7. The window is the same size with up to 4 active graphics and expands slightly with 5 or 6 graphics. The provision of 5 markers is very good but it should be possible to use less than that.

Thank you for all your work on this fantastic software.
73,
John

zarath, 2020-08-14:
@G4GCL don't know what about 5 markers? 
I initialize 3 if no .ini is present (eg. first installation)
Markers can be added or removed via the Display Setup window. - If there is a problem. That would be a new issue.

zarath, 2020-08-14:
Thiis bug will be fixed in v0.3.8 (it's fixed by https://github.com/NanoVNA-Saver/nanovna-saver/commit/6f0f55ab798e19a86a7603f3197351dd1061a609)

closed 2022-01-09 by zarath