NanoVNA-Saver/nanovna-saver #315: Number of points with N segments

igor-m, 2020-08-12:
There was a short discussion on nanovna-users group some time back whether the number of points with, say, 8 segments and 201sweep points is 1608 or 1601.
"OneOfEleven" (who closed his project and removed his files) indicated 1601 is the proper number - that what he used to use in his app.
Nanovna-saver calculates ie. 8*201=1608.
What is the correct number ? :)

PS: An attempt to save open source nanovna at https://groups.io/g/open-nanovna ..

[image]
[image]

zarath, 2020-08-13:
Who cares ....

MZachmann, 2020-08-13:
imho 1608 is clearly the right count. Anything else isn't very logical. There's no overlap thingy going on.

zarath, 2020-08-13:
I've missed that discussion but I've seen comments about 101 / 201 points, so that you go to the exact middle of a range... and with
adjacent starts and ends you get even numbers for the step size....
But with higher point count I don't see much sense in this...

DiSlord, 2020-08-14:
Point count depend from how program separate range on points:
OneOfEleven made for example this:
1-100MHz and 2 segments
so
1-50MHz (include 50Mhz)
50-100MHz (also include 50Mhz) - one point twice, so result not 201+201, result 200+201 = 401
For 4 segments 3 points measured twice, so not 201*4, result 201*4-3 = 801

zarath, 2020-08-14:
Totally understandable, but why not gain a little bit more resolution :-) And with current version (or V2 with 1023 point capability) it makes even less sense as one could do 51 point sweeps with eponential increasing step sizes between the segments.

closed 2020-11-01 by zarath