NanoVNA-Saver/nanovna-saver #316: Windows Version 0.3.6 Qt5Core.dll, Version: 5.15.0.0 crashed

Guesel, 2020-08-13:
If the number of segments  greater than 1 , the app crashed after a sweep starts. Windows reports an application-error:

[Report.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5067043/Report.txt)

Name der fehlerhaften Anwendung: nanovna-saver.exe, Version: 0.0.0.0, Zeitstempel: 0x5e11d35f
Name des fehlerhaften Moduls: Qt5Core.dll, Version: 5.15.0.0, Zeitstempel: 0x5eb90ecb
Ausnahmecode: 0xc0000409
Fehleroffset: 0x0000000000021478
ID des fehlerhaften Prozesses: 0x2d0c
Startzeit der fehlerhaften Anwendung: 0x01d671222065acaf
Pfad der fehlerhaften Anwendung: C:\VNA-SAVER\nanovna-saver.exe
Pfad des fehlerhaften Moduls: C:\VNA-SAVER\Qt5Core.dll

MZachmann, 2020-08-13:
Could you please run in debug mode and include the log output? To run in debug mode add   **-d debug.txt**    to the call and it will place debug output into the debug.txt file in the current folder. Thanks.

zarath, 2020-08-13:
it's `nanovna-saver.exe -D debug.txt` with a capital D ... small d dumps to console

Guesel, 2020-08-19:
The 0.3.7 does the job. thx

closed 2020-08-19 by Guesel