NanoVNA-Saver/nanovna-saver #320: NanoVNA-Saver 3.8 pre division by 0

PA3CCE, 2020-08-20:
This morning I started a fresh calibration in NanoVNA-Saver 0.3.8 pre, 1MHz to 2GHz, 10 Segm,ents, 401 datapoints, 100 Hz Bandwidth. The "short" caalibration went well, but while running the "open" calibration I got a "divide by "zero" errormessage.
See screenshot and errormessage "calibration.txt".

[image]

[calibration.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5102334/calibration.txt)

Jos

MZachmann, 2020-08-25:
When the sweep fails it returns a zero length array of data that causes a crash. The stack is shown in the calibration.txt.

zarath, 2022-09-29:
Zero length of array is checked in newer versions

closed 2022-09-29 by zarath