NanoVNA-Saver/nanovna-saver #322: NanoVNA-Saver 0.3.8 pre " Error during sweep"

PA3CCE, 2020-08-20:
After posting the previous message " division by 0 error" I restarted NanoVNA-Saver and tried again to do the same calibration, 1MHz to 2GHz, 10 Segments, 401 datapoints, 100 Hz Bandwidth. This time "short" and "open" went well, but during "load" calibration I got the errormessage  "Error during sweep" . See screenshot and debug file "calibraton_2.txt
Jos

[image]

[calibration_2.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5102470/calibration_2.txt)

PA3CCE, 2020-08-20:
After the above I restarted NanoVNA-Saver again in tried to do the caliubration again, this time I got the "Error during sweep" m,essage while I performed the "Isolation" sweep. 

[calibration_3.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5102545/calibration_3.txt)

Jos

PA3CCE, 2020-08-20:
Sorry but I must add that I use the by DiSlord adapted NanoVNA-H4.py and VNA.py files to make it possible to use all the bandwidth and number of datapoints settings that are in his latest firmware version 1.0.20.

zarath, 2020-08-21:
that firmware seems to be a little bit unreliable... as I see there were only 399 measurement points returned but 401 requested ...

RandMental, 2020-08-30:
Getting the same if I use the latest DiSlord H4 firmware

Testing V1.0.23  Firmware with Saver 0.3.8 Pre01 (as of today in the Development branch, 

When is select more than 1 Segment in saver, I get Error: list index out of range  (see below screen cut
[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5147504/debug.txt)
 .
  
[image]

C:\Python\Python38\python.exe -m NanoVNASaver.__main__ -D debug.txt
NanoVNASaver 0.3.8-pre01
 
Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors
 
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3
 
See https://github.com/NanoVNA-Saver/nanovna-saver for further details.
 
Settings: C:/Users/deond/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2020-08-30 23:12:20,191 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\SweepWorker.py", line 80, in run
    self._run()
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\SweepWorker.py", line 121, in _run
    freq, values11, values21 = self.readAveragedSegment(
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\SweepWorker.py", line 236, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\SweepWorker.py", line 264, in readSegment
    values11 = self.readData("data 0")
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\SweepWorker.py", line 279, in readData
    tmpdata = self.app.vna.readValues(data)
  File "C:\Workspaces\nanovna-saver\NanoVNASaver\Hardware\NanoVNA.py", line 129, in readValues
    f"{data[2]} {data[3]}"))
IndexError: list index out of range

[debug.txt](https://github.com/NanoVNA-Saver/nanovna-saver/files/5147507/debug.txt)

zarath, 2020-10-06:
Looks like a firmware issue...
As I doesn't have a -H4
If possible, can you try to insert a log statement like 'log.debug("Data %s", data)' on top of line 129 in NanoVNA.py?

Mike4U, 2020-10-19:
Finally figured out solution to this on NanoVNA-V2 with the latest version git-20201013-32077fd.
Before you start NanoVNA-Saver the NanoVNA-V2 must have averaging for slot 0 turned off.  You can do this without messing up the cal0 (I think) by:
display/avg/none
/back/back/calibrate/save/0
Turn of and on to verify 1x AVG at bottom of screen on startup.

jeanpierancori, 2022-02-08:
Hi friends,
does anyone have the installer for Nano VNA Saver 0.3.8?

closed 2022-09-21 by zarath