NanoVNA-Saver/nanovna-saver #325: Crystal measurement function

EB4APL, 2020-09-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

EB4APL, 2020-09-03:
Crystal measurement can be done with the current versions, but an automated function to directly obtain the motional parameters of a crystal or a set of them could be of interest.
I am used to do it with a Scalar Network Analyzer (PHSNA) which have this function and I have done some tests to automate and speed up the procedure. While I have collaborated in the later versions of PHSNA I am not a programmer and I don't know Phyton, but if any author likes the idea of implementing this functionality, please contact with me at eb4apl@gmail.com and I will supply him the methodology and the formulas.

stdevPavelmc, 2021-02-05:
Hi Igancio.

This is a interesting feature, can you provide me the methodology and the formulas to try to implement this feature into the software?

BTW: Soy cubano y podemos intercambiar en español internamente (I'm cuban and we can speak in spanish directly)

Please review your mail.

open-tommie, 2021-02-05:
Hi,

> 2021/02/06 1:48、Pavel Milanes (CO7WT) 
> Hi Igancio.
> 
> This is a interesting feature, can you provide me the methodology and the formulas to try to implement this feature into the software?
> 
I hope this video helps you.

Measuring Crystals with NanoVNA and other tools 
https://youtu.be/G9zZRNzhsEE

I want to measure crystals with NanoVNA-Saver.

From Tokyo, Japan.

stdevPavelmc, 2021-02-06:
Thanks @open-tommie, very instrutive!

@EB4APL please review the video and my notes below and comment, please...

Notes for later (may contain bugs)

- C0: Measured way below the original freq to remove the crystal, let's say 1Mhz below, then run a couple of sweeps to get a mean of that value. (Crystal in parallel to ch0, ch1 not used)
- Setup a fix to put the crystal in series with the signal of the NanoVNA (try to keep the 50 Ohms path matched!)
- Find the series and parallel resonance, set the center of the span in the middle point of both Fs & Fp
- Set a span that allows to view the entire slope of the curve with a safe margin (typical 50-150 kHz, this curve/settings can be used for later)
- Rm: Set a narrow span ~10khz around the series resonance frequency to determine Fs precisely
- Record S21 to find it
  -  Fs that is at S21 at 0 degrees phase (also at minimum s21 value)
- Compute Rm per the video/paper formulas
- G3UUR Oscillator method can't be done with the NanoVNA, or can we?
  - Reading the input and finding the peak at ch1, instructing the user to switch from trough the cap to directly to ground connection on the oscilator?
  - Level issues for DR at the input of the NanoVNA?
  - User formulas from the video/article
- Phase shift method is easier
  - Use the 10k max resolution centred over Fs (Record Fs)
  - Record S21 on Log Mag & Phase
  - Find Fl and Fh at +/- 45 degrees phase shift around Fs
  - Use formulas from video/article to calc Cm & Lm
- 3dB method is similar to Phase shift method, which give us better results?
- Series/Parallel method
  - Use the settings of the full view of the entire entire slope
  - Find rough Fs/Fp, make a closer 10kHz scan to take them precisely
  - Use the formulas and previous C0 values to calculate the values.

Some random thoughts:
- G3UUR is discarded as need an extra oscillator
- Use1/2 of F to check for the holder capacitance, or use a initial 1Mhz span and use the lower freq for that (as the NanoVNA must be calibrated at that frequency)
- 3dB & phase shift method can be carried out with a simple external fixture
- What method must we use, phase shift?
- This can be implemented easily after some effort.

Please review my notes and do comment on them... I will came up with a proposed algorithm for the users point of view for posting later on this weekend.

Cheers

vadim-dem, 2021-04-09:
[XQ_S11_Lm_calc_05.zip](https://github.com/NanoVNA-Saver/nanovna-saver/files/6287822/XQ_S11_Lm_calc_05.zip)
You can try also circle fitting into G/B plot, the attachment is Scilab script with .S2P file example:
[image]

closed 2023-03-02 by github-actions[bot]