NanoVNA-Saver/nanovna-saver #328: The 0.3.7 release of nanoVNA-saver cannot connect nanoVNA through bluetooth

I0KPT, 2020-09-20:
**Describe the bug**
Until today I used nanoVNA-saver 0.2.0 and I was able to connect it to my nanoVNA-H which is running a HC-05 Bluetooth module.  The OS is Win10. With 0.2.0 I had to manually insert the corresponding COM  (COM11 in my case) and everything was OK.

With the new 0.3.7 I am not able anymore to have nanoVNA-saver connected via Bluetooth to the nano.  I get this error message: 

2020-09-20 17:20:01,757 - NanoVNASaver.NanoVNASaver - ERROR - Tried to open None and failed: 'NoneType' object has no attribute 'open'

If I connect it using the USB cable, everything is ok.

Best regrads

Piero, I0KPT

igor-m, 2020-09-23:
Doublecheck the Bluetooth driver and it settings in Win10.
I've been using HC-05 with Win10 and Win7 almost daily and have problems with COM once a week :)
Also HC-05 here stops sending data (it goes into sleep mode) to the PC when none communication with the module for 5secs.
It wakes up upon receiving a byte from the PC. Note sure this could be your issue, however.

EU1KY, 2021-05-06:
The same issue appeared with EU1KY antenna analyzer that implements NanoVNA protocol on STM32 boards.
The solution was simple: to add STLink virtual COM port's VID/PID to USBDEVICETYPES in Hardware.py:
USBDEVICETYPES = (
    USBDevice(0x0483, 0x5740, "NanoVNA"),
    USBDevice(0x16c0, 0x0483, "AVNA"),
    USBDevice(0x04b4, 0x0008, "S-A-A-2"),
    **USBDevice(0x0483, 0x374B, "STLink")**
)

EU1KY, 2021-05-06:
Manual entering of the COM port in the dropbox, as it was in version 2, does not help: it does not create an interface from the port name. This must be fixed.

closed 2022-01-09 by zarath