NanoVNA-Saver/nanovna-saver #329: 0.3.7 NanoVNA Saver Unable to parse version: 0-2-1 with NanoVNA-F

cache4pat, 2020-09-24:
[image]

**Describe the bug**
0.3.7 NanoVNA Saver Unable to parse version: 0-2-1 with NanoVNA-F 

For the first time, I have hooked my new NanoVNA-F via my Windows 10 HP Envy (core i7) Com USB Port 4, to NanoVNASaver with version 0.3.7 installed.

The software connected, but then reported the ERROR shown in the attached image.

**Expected behavior**
Error free operation; data to appear on the graphic displays. Result .. Nothing  

**Screenshots**
See attached Image
[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] :HP Laptop running Windows 10 'HOME Edition' Version 1903 - See Images Provided
 - Python version: Version 3.8
 - NanoVNA-Saver version: [e.g. 0.2.1]

**Additional context**
Add any other context about the problem here.

tczerwonka, 2020-12-19:
I just encountered the same problem on Ubuntu 20.04, latest saver code from git.  I removed my NanoVNASaver.ini and the software continued and functioned as expected.  Not a fix, but maybe a workaround.

astrocfi, 2020-12-20:
I just ran into the same problem with 0.3.8 and the NanoVNA-F. I removed the NanoVNASaver.ini file as suggested by @tczerwonka, but it didn't help. I'm running Windows 10.

If you need help debugging this I'm happy to help. I'm an experienced Python developer.

DL1ABR, 2021-02-24:
astrocfi wrote and I have the same problem:
I just ran into the same problem with 0.3.8 and the NanoVNA-F. I removed the NanoVNASaver.ini file as suggested by @tczerwonka, but it didn't help. I'm running Windows 10.
Sometime Saver is running properly for some short time, then it quickly freezes - No Respond.
Restart Saver helps sometimes, sometimes not and Saver freezes at once after start. 
Restart of PC helps sometimes, sometimes not and Saver freezes at once after start.

DL1ABR, 2021-05-27:
I experienced as follows and SAVER is running much more stable:
I took a USB 2.0 port - not a USB 3.0 port.
I use the 32 bit version - not the 64 bit version.
Once if I run the 64 bit version, PC needs restart so that the 32 bit version is running.

closed 2022-11-10 by github-actions[bot]