NanoVNA-Saver/nanovna-saver #334: Sweep Generator: Offset frequency

sbridger, 2020-10-05:
Receiver tuning uses a sweep generator. However the input frequency is not the same as the output frequency. 
You often want to look at the IF frequency (or the RX output) while sweeping the input frequency. These are usually a narrow band sweep - approx the IF width,

**Describe the solution you'd like**
a) An offset frequency between output and input.
b) for very narrow band sweeps, the rate may need to be slowed.

beyond this it might be useful to have non-RF input, but I imagine this is also a firmware/hardware change..
c) Audio or DC input

zarath, 2020-10-07:
Sorry, I think this will be impossible, as all mixers are connected to one oscillator in the hardware...
Consider buying two tinaSA's ...

closed 2020-10-07 by zarath

sbridger, 2020-10-07:
[V2 appears to use separate oscillators](https://s3-us-west-1.amazonaws.com/groupsioattachments/34671/34272222/2947/0?AWSAccessKeyId=AKIAJECNKOVMCCU3ATNQ&Expires=1602102977&Signature=zgZr2D61ickhlbM0aWSUO4BQvrs%3D&response-content-disposition=inline%3B+filename%3D"Preliminary+Block+Diagram.png")

zarath, 2020-10-08:
Even then the firmware has to implement it first