NanoVNA-Saver/nanovna-saver #337: Fundamental Calculation Flaw Major Fix

FROSTBITEDONKEY, 2020-10-18:
**Describe the bug**
Conversion to dB for S parameter error. This makes tool useless for S11 AND S21. 
 
Tool uses 10log10(#), s-parameters are in voltage. Tool should use 20log10(#) to convert into dB and 10^(#/20) to convert from dB. 

**To Reproduce**
Steps to reproduce the behavior:
1) Save any snp file and perform the math
2) load snp file that is in dB format to display in application. All numbers displayed in application in dB format are half the value they should be. 

**Expected behavior**
Tool should use 20log10(#) to convert into dB and 10^(#/20) to convert from dB. 

Performed on Windows 10 machine.

Goodluck with the fix. Love this application! Just want it to produce correct values.

zarath, 2020-10-25:
I'm wondering how you have saved snp in dB-Format.

I just doing MA-Format on saving.
When loading I do 10^(#/20) 
see
https://github.com/NanoVNA-Saver/nanovna-saver/blob/6b554dca107037ab46c7f3f078008c9389c9735a/NanoVNASaver/Touchstone.py#L197

FROSTBITEDONKEY, 2020-10-26:
I want to say that you've done a brilliant job with this tool and the
interface. Really great stuff! So much so that I wanted to use it as a data
viewer to compare traces with ones taken on another Network analyzer. My
problem has been with the viewing side. This would really stand out if you
can add multiple 'reference' traces.

That's when I loaded up a touchstone file that was in dB mag angle format
to see what nanovna-saver had prebuilt for it's touchstone import since
they are marked in the RI format.

I was surprised to see that it actually took the dBm value file without
throwing a flagg but produced weird numbers. I tried looking for the code
but was unsuccessful. Thank you for highlighting the code to show the
conversion from the text vile assuming voltage (using 20 log instead of 10
log)!

I guess the issue is that the user , myself in this case, did not
understand that it only took RI instead of the other formats, and assumed
that there was an issue with the calculation.

Maybe a solution could be to flagg a touchstone files if it's not in a
format that you accept? OR to traverse the last comment line of a
touchstone file and adjust the import accordingly? Or both? Sorry I that's
just more work. :)

Thank You Again!

Cam
_____________________________
Cameron Weir
RF/Electronics Engineer

zarath, 2020-10-26:
can you send me such a file? are there multiple comments in it?

closed 2022-11-10 by github-actions[bot]