NanoVNA-Saver/nanovna-saver #338: 38-pre release

DG9BFC, 2020-10-18:
i have used 38 pre 03 and it works fine (16 days old)
then i tested last version (8 days old) and it does not start
system used is a win7 64 bit on a core2duo
cause it did not start here (while other worked fine) i did not try it on my win10 tablet yet
greetz sigi dg9bfc

DG9BFC, 2020-10-28:
[image]
debug text did not work so i made a screenshot

DG9BFC, 2020-10-28:
tested bump version ... does not work
tested rc version ... does not work
but the version a few days before the bump version works fine (38pre03)

DG9BFC, 2020-10-28:
[image]
this version is just a few days before the bump version and it works fine

DG9BFC, 2020-10-28:
and here debug from the working version
.....................................copy.........................................
2020-10-28 17:59:39,457 - NanoVNASaver - INFO - Startup...
2020-10-28 17:59:39,488 - NanoVNASaver.NanoVNASaver - DEBUG - Running from pyinstaller bundle
2020-10-28 17:59:39,504 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-28 17:59:39,504 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-10-28 17:59:39,504 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-28 17:59:39,504 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-10-28 17:59:39,519 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-10-28 17:59:39,925 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-10-28 17:59:39,972 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-10-28 17:59:40,783 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-10-28 17:59:41,017 - NanoVNASaver.Hardware.Hardware - DEBUG - Found S-A-A-2 USB:(04b4:0008) on port COM32
2020-10-28 17:59:41,032 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-10-28 17:59:43,918 - NanoVNASaver.NanoVNASaver - INFO - Connection COM32 (S-A-A-2)
2020-10-28 17:59:44,106 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-10-28 17:59:44,324 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-V2
2020-10-28 17:59:44,386 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - readVersion: 1.0.2
2020-10-28 17:59:44,386 - NanoVNASaver.Hardware.VNA - DEBUG - Features: {'Customizable data points', 'Multi data points'}
2020-10-28 17:59:44,511 - NanoVNASaver.NanoVNASaver - INFO - Read starting frequency 200000000 and end frequency 300000000
2020-10-28 17:59:44,511 - NanoVNASaver.NanoVNASaver - DEBUG - Starting initial sweep
2020-10-28 17:59:44,511 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-10-28 17:59:44,511 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(200000000, 300000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-28 17:59:44,511 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (200000000, 300000000)
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 0
2020-10-28 17:59:44,511 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (200000000, 300000000)
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - INFO - Reading from 200000000 to 300000000. Averaging 1 values
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 200000000 to 300000000
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Read 101 frequencies
2020-10-28 17:59:44,511 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-10-28 17:59:44,652 - NanoVNASaver.Hardware.NanoVNA_V2 - INFO - reading values
2020-10-28 17:59:46,773 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Freq index from: 5
2020-10-28 17:59:46,773 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Freq index to: 4
2020-10-28 17:59:46,789 - NanoVNASaver.SweepWorker - DEBUG - Read 101 values
2020-10-28 17:59:46,789 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
2020-10-28 17:59:46,789 - NanoVNASaver.SweepWorker - DEBUG - Read 101 values
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 101, Offset: 0
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - Saving data to application (101 and 101 points)
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-10-28 17:59:46,804 - NanoVNASaver.SweepWorker - DEBUG - Sending "finished" signal
...............................................................end copy............debug.txt

zarath, 2020-10-30:
Ok. That was a problem cause by pyinstaller in .exe generation. I have updated the installer from 3.6 to 4.0 in github actions.
Should work now.

DG9BFC, 2020-10-30:
just installed 038rc02 and it works so issue can be closed
guessing you can relase it to public (means put on main page so other users find it without searching in the actions tab)

DG9BFC, 2020-10-30:
and i made a "crosscheck" ... it also works with the fw from ojisan
.....................copy debug txt w ojisan fw....................................
2020-10-30 21:49:18,996 - NanoVNASaver - INFO - Startup...
2020-10-30 21:49:19,058 - NanoVNASaver.NanoVNASaver - DEBUG - Running from pyinstaller bundle
2020-10-30 21:49:19,074 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-30 21:49:19,074 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-10-30 21:49:19,074 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-30 21:49:19,074 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2020-10-30 21:49:19,074 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-10-30 21:49:19,448 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2020-10-30 21:49:19,526 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2020-10-30 21:49:19,869 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 164.000000, normal font: 164.000000, factor: 1.000000
2020-10-30 21:49:20,025 - NanoVNASaver.Hardware.Hardware - DEBUG - Found S-A-A-2 USB:(04b4:0008) on port COM32
2020-10-30 21:49:20,041 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface
2020-10-30 21:49:26,640 - NanoVNASaver.NanoVNASaver - INFO - Connection COM32 (S-A-A-2)
2020-10-30 21:49:26,827 - NanoVNASaver.Hardware.Hardware - INFO - Finding correct VNA type...
2020-10-30 21:49:27,045 - NanoVNASaver.Hardware.Hardware - DEBUG - Retry detection: 1
2020-10-30 21:49:27,326 - NanoVNASaver.Hardware.Hardware - INFO - Type: NanoVNA-V2
2020-10-30 21:49:27,389 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - readVersion: 1.0.1
2020-10-30 21:49:27,389 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Hack for s21 oddity in first sweeppoint
2020-10-30 21:49:27,389 - NanoVNASaver.Hardware.VNA - DEBUG - Features: {'S21 hack', 'Multi data points', 'Customizable data points'}
2020-10-30 21:49:27,513 - NanoVNASaver.NanoVNASaver - INFO - Read starting frequency 200000000 and end frequency 300000000
2020-10-30 21:49:27,513 - NanoVNASaver.NanoVNASaver - DEBUG - Starting initial sweep
2020-10-30 21:49:27,513 - NanoVNASaver.NanoVNASaver - DEBUG - Starting worker thread
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2020-10-30 21:49:27,513 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(200000000, 300000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2020-10-30 21:49:27,513 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (200000000, 300000000)
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Sweep segment no 0
2020-10-30 21:49:27,513 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (200000000, 300000000)
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - INFO - Reading from 200000000 to 300000000. Averaging 1 values
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Reading average no 1 / 1
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Setting sweep range to 200000000 to 300000000
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Read 101 frequencies
2020-10-30 21:49:27,513 - NanoVNASaver.SweepWorker - DEBUG - Reading data 0
2020-10-30 21:49:27,654 - NanoVNASaver.Hardware.NanoVNA_V2 - INFO - reading values
2020-10-30 21:49:29,151 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Freq index from: 0
2020-10-30 21:49:29,151 - NanoVNASaver.Hardware.NanoVNA_V2 - DEBUG - Freq index to: 101
2020-10-30 21:49:29,183 - NanoVNASaver.SweepWorker - DEBUG - Read 101 values
2020-10-30 21:49:29,183 - NanoVNASaver.SweepWorker - DEBUG - Reading data 1
2020-10-30 21:49:29,183 - NanoVNASaver.SweepWorker - DEBUG - Read 101 values
2020-10-30 21:49:29,183 - NanoVNASaver.SweepWorker - DEBUG - Averaging 1 values
2020-10-30 21:49:29,198 - NanoVNASaver.SweepWorker - DEBUG - Calculating data and inserting in existing data at index 0
2020-10-30 21:49:29,198 - NanoVNASaver.SweepWorker - DEBUG - update Freqs: 101, Offset: 0
2020-10-30 21:49:29,214 - NanoVNASaver.SweepWorker - DEBUG - Saving data to application (101 and 101 points)
2020-10-30 21:49:29,214 - NanoVNASaver.SweepWorker - DEBUG - Sending "updated" signal
2020-10-30 21:49:29,214 - NanoVNASaver.SweepWorker - DEBUG - Sending "finished" signal
.....................................................end copy...........................................................

so saver 38 works with both fw ... time to make the next release??

closed 2020-10-30 by DG9BFC