NanoVNA-Saver/nanovna-saver #339: Support for NanoVNA SAA-2N

Allasso, 2020-10-22:
It appears that the SAA-2N is not on the list for support.  However, according to this video, the presenter has gotten NanoVNA-saver to work with the SAA-2N.

https://www.youtube.com/watch?v=5rMrRQjGfS4

Referring to connecting to the NanoVNA, the presenter says, "connect the device under com port-5, no problems whatsoever" (at 3:00)

Allasso, 2020-10-22:
(Didn't finish the original comment, apparently github recognizes multiple return key presses as "submit comment")

I'm trying out a new SAA-2N.  I am running NanoVNA-saver on Windows, and have no problem connecting to my old NanoVNA.  I plugged the SAA-2N model and Windows recognized it automatically ran the new device setup, setting it up on port 7.  However NanoVNA-saver does not recognize the port for the SAA-2N, after rescanning, and restarting both the app and the device.

If I can believe the video presenter, there apparently is some way to get this to work.

Allasso, 2020-10-27:
In addition, it was suggested that I verify I have the Cypress driver installed, which I do and in use for the V2.  I installed QT as well, and it communicates with the V2 just fine so I'm pretty confident it is not an issue with the device.

zarath, 2020-10-30:
#343

closed 2020-10-30 by zarath