NanoVNA-Saver/nanovna-saver #340: V0.3.7 x64  "Manage Bands, only input until 2100000000Hz. For QO100 2400000000Hz not possible"

DL9NAZ, 2020-10-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.1.4]

**Additional context**
Add any other context about the problem here.

zarath, 2020-11-01:
done in v0.3.8

closed 2020-11-01 by zarath

DL9NAZ, 2020-11-02:
Danke für die Antwort,

habe es leider zu spät gesehen.

73 de Knut

PR #341 (2020-10-25): Development
Fix #340 340