NanoVNA-Saver/nanovna-saver #343: NanoVNA-saver does not recognize serial port for SAA2N

Allasso, 2020-10-27:
I originally filed this as a feature request as I originally thought the SAA2N may not be supported, but after discussion from other groups I believe that NanoVNA-saver should support the SAA2N, so I am filing this as a bug.

**Describe the bug**

NanoVNA-saver does not recognize serial port for SAA2N

**To Reproduce**
1. Plug SAA2N into USB port
2. Turn on SAA2N
3. Open NanoVNA-saver
4. Observe "Serial port" list under "Serial port control" section
5. Observe no ports are listed
6. Click "Rescan"
7. Observe still no ports are listed
8. Click "Connect to NanoVNA"
9. Observe no connection is made

**Expected behavior**
1. After performing steps 1 - 4 or 1 - 6 above, should see serial port listed for the SAA2N
2. Click "Connect to NanoVNA"
3. Connection is made to SAA2N

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows 8
 - Python version:
 - NanoVNA-Saver version: 0.2.2-1

**Additional context**

I have no problem connecting to my old NanoVNA.

I have verified that the Cypress driver is installed (as suggested by someone,) which I do and is in use for the V2.

I installed QT and it communicates with the V2 just fine so I'm pretty confident it is not an issue with the device.

zarath, 2020-10-27:
@Allasso NanoVNA-Saver version: 0.2.2-1 really?

Support for SAA2 is only in Versions above > 0.3.x

Allasso, 2020-10-27:
Thanks.  I'll update and report back.  Seems like I only installed saver a couple of months ago, but my mind may be slipping on that.

Allasso, 2020-10-27:
Okay, that did the trick.  Thank you and sorry for not checking the obvious.

closed 2020-10-27 by Allasso

zarath, 2020-10-28:
Have fun!

Allasso, 2020-10-30:
Well sorta - Now I'm running into #277 and #322 et al :-/  I got word that you're on the case, so I'll just wait it out.

Sorry for forgetting to close #339 btw.