NanoVNA-Saver/nanovna-saver #346: Division by zero error

Wojtus11, 2020-11-01:
**Describe the bug**
nanovna-saver crashes rarely with "Division by zero" error. Probably some communication error with device followed by lack of input checking. Encountered several times during continuous sweep (but it is probably not connected to sweep mode).

**To Reproduce**
Steps to reproduce the behavior:
1. Run continuous sweep
2. Wait... unfortunately this is rare and random.

**Expected behavior**
"Soft" error, forced disconnection of device or, in the best case, automatic resweep.

**Console dump**
2020-11-01 18:11:36,076 - NanoVNASaver.Hardware.NanoVNA_V2 - WARNING - expected 3232 bytes, got 0
2020-11-01 18:11:39,716 - NanoVNASaver.SweepWorker - ERROR - division by zero
Traceback (most recent call last):
  File "NanoVNASaver\SweepWorker.py", line 80, in run
  File "NanoVNASaver\SweepWorker.py", line 121, in _run
  File "NanoVNASaver\SweepWorker.py", line 253, in readAveragedSegment
  File "<__array_function__ internals>", line 5, in average
  File "numpy\lib\function_base.py", line 381, in average

**Desktop (please complete the following information):**
 - Windows7/x64
 - NanoVNA-Saver version: 0.3.8

**Additional context**
Encountered with v2 from HCXQS.

DG9BFC, 2020-11-05:
[image]

DG9BFC, 2020-11-05:
i see also the divide by zero error if i have avg switched on in the v2
when i switch avg off before connecting to device then safer works

DG9BFC, 2020-11-07:
it is a timing issue so please increase the timeout timer so that any avg setting can be used without crash or error ...

jkenny23, 2020-12-10:
Just a note I am also seeing this error with a NanoVNA v2 on Windows 10 (1809) with version 0.3.8. I might give a stab at fixing this (at least catching the exception so it goes from a hard crash to a soft one; it's annoying having to restart the program and restore calibration when this occurs).

DG9BFC, 2020-12-11:
Use newer fw (newest) And set avg to none and save in slot 0 before connecting to saverIf as example you set avg to x5Soft asks for 100pt but gets 20 pt backDg9bfc sigi

jkenny23, 2020-12-11:
Do you know where the timeout timer is in the code? I would like to keep the averaging feature on. The incidence is lower with the newer firmware I believe (happened about 2-3 times in 600 sweeps yesterday, after firmware update it crashed after about 1500 sweeps). I've modified the program to take a sweep every ~10s to torture test the connection.

DG9BFC, 2020-12-11:
Avg none only needed in usage with saver.. And in saver you can then add avg inside pc soft..And as far as i know there is an experimental fw that does not crash even if avg is not set to 0Dg9bfc sigi

jkenny23, 2020-12-11:
Not sure I understand your meaning, I double checked my VNA config, it is set to "None" on the device for averaging. The averaging I have enabled is the default 3 sweep average in NanoVNA-Saver "3/0".

PR #386 (2021-06-22): Pull/mag loop analysis
Now MagLoop Analysis change start/stop frequency and run sweep, so you can zoom on minimum.
If "run automatically" is flagged, it keep running, so you can tune your magloop
( if you can see nanovna-saver ;-) )
To stop, remove "run automatically" flag.
Todo
more smart change to frequency
For keep running without errors I had to:

- change vswr when R<0
- retry reading if non results. Fix #346 ?
closed 2022-09-18 by zarath