NanoVNA-Saver/nanovna-saver #347: Sweep range reset on connect

Wojtus11, 2020-11-01:
**Describe the bug**
Every time device is connected, sweep range is reset to 200-300MHz.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect device
2. Set sweep range 10-100MHz (or any other)
3. Disconnect device
4. Connect device

**Expected behavior**
Sweep range as input should be kept.

**Desktop (please complete the following information):**
 - Windows 7/x64
 - NanoVNA-Saver version: 0.3.8

**Additional context**
Encountered with v2 from HCXQS

galileo-pkm, 2020-11-02:
For me (Linux, NanoVNA-H) it always resets to 27-30MHz. 
Also resets the datapoints and bandwith. Quite annoying.

DG9BFC, 2020-11-09:
it is cause on your both systems no inifile can be found from the software

galileo-pkm, 2020-11-09:
No, inifile is there, populated and used.

Wojtus11, 2020-11-09:
My ini is at C:\Documents and Settings\User\AppData\Roaming\NanoVNASaver\NanoVNASaver.ini
It's populated and written by Saver. There are no sweep limits inside, however.

gaionim, 2020-12-23:
sweep is hardcoded in NanoVna.init()

...
        self.start = 27000000
        self.stop = 30000000
...

If I change in 

        frequencies = self.readValues("frequencies")
        self.start = frequencies[0]
        self.stop = frequencies[-1]

read sweep from device.
Seems to work on my old NanoVna, but I do not know on other !!!

zarath, 2020-12-23:
won't work on V2

gaionim, 2020-12-23:
can we override self.readValues("frequencies") with the correct command ?
I can't use self.readFrequencies() because it use "scan", which need start,stop we want read :-(
I can define function ( es.  _read_frequencies ) only for V1 and fallback to 27M,30M.

gaionim, 2020-12-23:
See pull request and check if if works on V2 please.

galileo-pkm, 2021-01-08:
This seems to be firmware dependent. Just got a H4 and with stock firmware
and upon start the range was set to 50KHz-900MHz. Once I updated to Dislords firmware
the range gets set to 27-30.

gaionim, 2021-01-08:
forgive for my english ...
It is because different method
old,   
sweep {start} {stop}  -> set range  only one time if not changes
data 0  -> every time read last data
new
scan {start} {stop} -> read data in range 

In my pull request ( now merged in development ) ONLY FOR NANOVNA ( v1 ) because seems not to work on v2 
try to read frequencies old style even if firmware is capable of scan_mask 
Only if some error fallback to default  27-30

I can't test on other than v1 hardware, but i can try to  move NanoVNA._get_running_frequencies() to VNA without check name and fallback if some error

galileo-pkm, 2021-01-08:
I'm not quite sure if I understood you correctly but I tried the development branch
and the behaviour is the same.
This is with a V1 H4 (H4 is the Hugens 4" screen version).

With a 2.8" device (same Dislord 1.0.45 firmware) the range is set
to 50k-900M. 
So it seems that there is a difference between firmware versions.

gaionim, 2021-01-08:
try new pull request
https://github.com/NanoVNA-Saver/nanovna-saver/pull/368

galileo-pkm, 2021-01-08:
It works. Tested with a "4 and a 2.8" device.
Thanks.
ATM only feature that is missing is the 401 sweep points #367 
Maybe an easy fix?

gaionim, 2021-01-08:
glad to hear that;
for 401 sweep points I do not have hardware to try :-(

galileo-pkm, 2021-01-08:
It turned out as an easy fix, I have updated the issue with a pull request.

closed 2022-10-07 by zarath