NanoVNA-Saver/nanovna-saver #350: NanoNVA keeps freezing

Double00Dalo, 2020-11-17:
**Describe the bug**
A clear and concise description of what the bug is.
Whenever I click to connect to my nanovna, it freezes and stops responding
**To Reproduce**
Steps to reproduce the behavior:
1. Open nanovnasaver
2. Click on connect to device
4. See error

**Expected behavior**
I expected it to continue normally and show graphs from my nanovna on the software.

**Screenshots**
Not exactly sure how to do this

**Desktop (please complete the following information):**
 - OS: [Windows 10}
 - NanoVNA-Saver version: [e.g. 0.3.8]

**Additional context**
Add any other context about the problem here.

DL1ABR, 2021-02-24:
As astrocfi wrote and I have the same problem:
I just ran into the same problem with 0.3.8 and the NanoVNA-F. I removed the NanoVNASaver.ini file as suggested by @tczerwonka, but it didn't help. I'm running Windows 10.
Sometime Saver is running properly for some short time, then it quickly freezes - No Respond.
Restart Saver helps sometimes, sometimes not and Saver freezes at once after start.
Restart of PC helps sometimes, sometimes not and Saver freezes at once after start.

zarath, 2022-05-20:
Outdated Version

closed 2022-05-20 by zarath