NanoVNA-Saver/nanovna-saver #362: Seems to have broken with OSX Big Sur

bwalker000, 2020-12-27:
Upon launch I had the system lock with the following message. This is true for both versions 0.3.7 and 0.3.8

This software worked well for me a couple of months ago. The only change I can see being my update to Big Sur. OSX 11.1

---------------------------------------------------------------------------------------------------------------------------

NanoVNASaver 0.3.8

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: /Users/Bill/.config/NanoVNASaver/NanoVNASaver.ini
2020-12-27 09:39:13,260 - NanoVNASaver.Windows.About - ERROR - Checking for updates produced a URL exception: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1124)>
Traceback (most recent call last):
  File "urllib/request.py", line 1350, in do_open
  File "http/client.py", line 1255, in request
  File "http/client.py", line 1301, in _send_request
  File "http/client.py", line 1250, in endheaders
  File "http/client.py", line 1010, in _send_output
  File "http/client.py", line 950, in send
  File "http/client.py", line 1424, in connect
  File "ssl.py", line 500, in wrap_socket
  File "ssl.py", line 1040, in _create
  File "ssl.py", line 1309, in do_handshake
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1124)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "NanoVNASaver/Windows/About.py", line 158, in findUpdates
  File "urllib/request.py", line 222, in urlopen
  File "urllib/request.py", line 525, in open
  File "urllib/request.py", line 542, in _open
  File "urllib/request.py", line 502, in _call_chain
  File "urllib/request.py", line 1393, in https_open
  File "urllib/request.py", line 1353, in do_open
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1124)>

--------------------------------------------------------------------------------------------------------------------------

I'm not clever enough to know what to do about this.

Running a 15" Macbook Pro mid 2015
Quad Core Intel i7
16 GB Memory

felipe3d, 2020-12-28:
The same is happen with me

jcw, 2021-01-02:
My install is not even reaching that point. After a minute, the first msg appears, and the last one is "Settings: ...".

jcw, 2021-01-02:
Ok, some progress, with `./nanovna-saver -d` the last line shown is:

    2021-01-02 14:25:11,863 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface

Serial USB connnection seems to be ok, but no GUI window ever appears.

bwalker000, 2021-04-21:
https://github.com/NanoVNA-Saver/nanovna-saver/issues/380#issuecomment-792289753

closed 2022-10-07 by zarath