NanoVNA-Saver/nanovna-saver #367: NanoVNA H-4, 401 sweep points

galileo-pkm, 2021-01-08:
The Nano H4, due to larger memory, has support for 401 sweep points.
Saver does not seem to detect that. Can this be enabled?
This is with Dislords 1.45 firmware.

galileo-pkm, 2021-01-08:
It appears to be an easy fix, (valid_datapoints = (101, 11, 51, 201, 401) in NanoVNA_H4.py)
Pull request: https://github.com/NanoVNA-Saver/nanovna-saver/pull/369

jkrnl, 2021-02-21:
Solution seems not to be stable. Occasionally throws error what dissapears when I switch back to 101
Please, could you check it?

`2021-02-21 15:17:40,290 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\SweepWorker.py", line 80, in run
    self._run()
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\SweepWorker.py", line 121, in _run
    freq, values11, values21 = self.readAveragedSegment(
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\SweepWorker.py", line 236, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\SweepWorker.py", line 264, in readSegment
    values11 = self.readData("data 0")
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\SweepWorker.py", line 279, in readData
    tmpdata = self.app.vna.readValues(data)
  File "c:\users\x\appdata\local\programs\python\python39\lib\site-packages\NanoVNASaver\Hardware\NanoVNA.py", line 147, in readValues
    f"{data[2]} {data[3]}"))
IndexError: list index out of range`

galileo-pkm, 2021-02-21:
Your firmware version?

jkrnl, 2021-02-21:
> Your firmware version?
sorry:
DiSlord 1.0.45
MaybeI found a discrepancy on bandwidth selection, this fw offers :30/100/333/1000/2000 and 4000 Hz on device, but not all of these are shown on NanoVNA saver dropdown.
Thank you

galileo-pkm, 2021-02-21:
I have been using the same firmware and haven't seen any issues
but I rarely use anything other than 333Hz BW.
Anyway this change adds only a single array element, no actual code change
so the issue is somewhere else.

closed 2022-11-10 by github-actions[bot]