NanoVNA-Saver/nanovna-saver #379: Weird behavior of offset delay

mcesar-rlacruz, 2021-03-02:
Hi,

Just arrived here, so maybe it's simply me not really knowing what I'm doing, but I found a strange behavior of the "offset delay" option in the calibration menu. What I expect varying this parameter is to compensate for different cable/port-extension lengths, so I expect the traces in the Smith chart to rotate and also to expand/compress, but not distort. But what I find is that modifying the _offset delay_ parameter the trace is heavily distorted (see below).

**Describe the bug**
Parameter _offset delay_ modifies the Smith chart traces in an unexpected way.

**To Reproduce**
Steps to reproduce the behavior:
1. Calibrate (SOL) your nanovna and store the calibration in slot0
2. Open nanovna-saver
3. Adjust the sweep parameters (in the example below it is 301 points from 2.3 GHz to 2.6 GHz)
4. Perform a sweep
5. Open the calibration menu
6. Reset the calibration
7. Insert the Short. Perform a sweep and push button "Short"
8. Repeat above step with both O and L
9. Check that, now, the three S-O-L are measured as expected (dots at left, right and center of the Smith chart)
10. Insert a small port extension and the S or O (in the example below I used a SMA M-F barrel and the O)
11. Perform a sweep
12. Modify the "offset delay" parameter trying to mutate the trace into a point at Z=∞, the trace is heavily distorted

**Expected behavior**
There must be a value of "offset delay" that moves the trace into a single dot at Z=∞ (or 0 if you use the S)

**Screenshots**
This is the O measured after calibration (markers are at 2.4, 2.45 and 2.5 GHz, start at 2.3 GHz, end at 2.6GHz)
[image]

This is the O with a small extension
[image]

And this is the O with a small extension and a value for "offset delay" of -60.5 ps. When measuring directly on the nanoVNA (w/o nanovna-saver) a value of 2·60.5ps = 121 ps for "electrical delay" converts the trace above into a point at Z=∞
[image]

**Desktop (please complete the following information):**
 - OS: Win10 64 Enterprise
 - Python version: Python 3.8.3 (for nanovna-saver 0.3.9-pre)
 - NanoVNA-Saver version: 0.3.9-pre (Python) and 0.3.8 (precompiled for Win)
 - NanoVNA V2+4

And that's all. Thanks a lot for this superb work and regards.

jamesl-dm, 2021-07-20:
I agree - changing the offset delay should change the phase of all measurements, but shouldn't change the amplitude. The return loss graph should be completely unaffected by the offset delay, but it currently isn't.

jamesl-dm, 2021-07-20:
Seems like it doesn't change the amplitude if there is no calibration set, so its some kind of interaction between the calibration and the offset delay.

jamesl-dm, 2021-07-20:
Looks like the code applies the cable correction first, before applying the calibration.
Shouldn't it apply the calibration first, to nail down exactly what was seen at the calibration plane.
Then extrapolate from the calibration plane to the end of the cable afterwards?

jamesl-dm, 2021-07-20:
Added PR #419