NanoVNA-Saver/nanovna-saver #380: 0.3.8 hang on macOS Big Sur at startup

cjsmithuk, 2021-03-07:
NanoVNA-saver 0.3.8 is hanging at startup on Big Sur on M1 Mac Mini. I've reproduced it successfully on an Intel iMac as well.

CPU is rammed at 100% on one core. The following is observed in the debug log:

`% ./nanovna-saver -d
NanoVNASaver 0.3.8

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

2021-03-07 14:29:39,490 - NanoVNASaver - INFO - Startup...
2021-03-07 14:29:42,576 - NanoVNASaver.NanoVNASaver - DEBUG - Running from pyinstaller bundle
Settings: /Users/chris/.config/NanoVNASaver/NanoVNASaver.ini
2021-03-07 14:29:44,641 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2021-03-07 14:29:44,641 - NanoVNASaver.SweepWorker - INFO - Initializing SweepWorker
2021-03-07 14:29:44,641 - NanoVNASaver.Settings.Sweep - DEBUG - Sweep(3600000, 30000000, 101, 1, Properties('', SweepMode.SINGLE, (3, 0), False))
2021-03-07 14:29:44,641 - NanoVNASaver.Settings.Sweep - DEBUG - get_index_range(0) -> (3600000, 30000000)
2021-03-07 14:29:44,642 - NanoVNASaver.SweepWorker - DEBUG - Init data length: 101
2021-03-07 14:29:44,738 - NanoVNASaver.NanoVNASaver - DEBUG - Building user interface
2021-03-07 14:29:44,804 - NanoVNASaver.Windows.About - DEBUG - Starting timer
2021-03-07 14:29:44,810 - NanoVNASaver.Windows.SweepSettings - DEBUG - update_band(False)
2021-03-07 14:29:44,826 - NanoVNASaver.NanoVNASaver - DEBUG - New font width: 137.500000, normal font: 137.500000, factor: 1.000000
2021-03-07 14:29:44,862 - NanoVNASaver.NanoVNASaver - DEBUG - Finished building interface`

Looking at a process sample from the OS, it's stuck in the bowels of Qt somewhere...

[image]

Assuming this is a Qt issue rather than a NanoVNA-saver issue based on the process sample but I'm not sure how much of Qt is shipping with the binary provided.

cjsmithuk, 2021-03-07:
Replying with workaround. You can issue the following at startup and it'll work, until Qt is fixed.

```QT_MAC_WANTS_LAYER=1 ./nanovna-saver```

zarath, 2022-01-09:
Duplicate of #362

closed 2022-01-09 by zarath