NanoVNA-Saver/nanovna-saver #393: Use as a frequency generator.

ortegahernandes, 2021-05-06:
Currently the "nanovna saver" does not allow to put the stimulus (start and stop) via PC on the same frequency
EX:
start: 433Mhz
stop: 433Mhz.

However, in the device it is possible to make such a configuration and use it as a frequency generator, only a specific frequency instead of a scan.

ok1vaw, 2021-05-20:
The CW generator function is only available from the Nanovna (V2) itself. The  V2+4 unfortunately has not the CW implemented.

ortegahernandes, 2021-05-20:
> The CW generator function is only available from the Nanovna (V2) itself. The V2+4 unfortunately has not the CW implemented.
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
English is not my native language, so I may not have understood it well.
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
My nano vna - H4 has this function (CW). just put the same frequency in start and stop.

How far do I know the H4 version is the (V1) correct?

is running the latest available firware.

[](https://github.com/hugen79/NanoVNA-H/releases/tag/1.0.45)

only that the nanosaver software does not allow to place the same frequency.

zarath, 2022-01-04:
As it depends too much on type of device and firmware I see no value in implementing this.

closed 2022-01-04 by zarath

ok1vaw, 2022-01-04:
Using NanoVNA (V2) as a CW generator is a very usefull function and its enhancing its value for RF measurement. I vote for leaving it there, where it works.