NanoVNA-Saver/nanovna-saver #395: Calibrate manual failure

IZ1MDJ, 2021-05-16:
**Describe the bug**
Calibrate in manual mode , I get the error :
Two of short open and load returned the same values at frequency 50000Hz

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Calibration'
2. Connect the short  Click on 'Short'
3. Connect  the open  Click on  'Open'
4. Connect the load   Click on   'Load'
5. Connect TX RX using short cable Click on 'Through'
6. Connect load to TX RX  Click on 'isolation'
Screen shoot
[image]

7. Click on 'Apply'

Screen shoot

[image]

**Desktop (please complete the following information):**
 - OS: win7 64bit Home Premium
 - Python version: [if applicable]
 - NanoVNA-Saver version: v.0.3.8
 
 If I use the calibration assistant all is working well.
 The nanovna was calibrate standalone , same frequency range and number of points

va3fod, 2021-05-20:
> **Describe the bug**
> Calibrate in manual mode , I get the error :
> Two of short open and load returned the same values at frequency 50000Hz
>
> **To Reproduce**
> Steps to reproduce the behavior:
>
> 1. Go [...]

I am having the same issue (NanoVNA-F V2 4.3 Inch)

PA3CCE, 2021-06-24:
I had the same problem while using version 0.3.9  !!
A stand-alone calibration of the device and saving it in pos 0 solved the problem

Wojtus11, 2021-12-14:
Guys, it is required to re-sweep before pressing "Short", "Open" etc. buttons. They only save data from last sweep as calibration. It becames obvious if you notice that they do it in no-time.

nightmechanic, 2021-12-27:
I am having the same issue,
either manual cal with separate sweeps for each standard,
or using the cal assistant:

I get the "Two of... " error when trying to apply the calibration.

The device has a fresh standalone calibration stored in slot 0.

OS: Mac OS 12.1 (also happened on 11.x)
Python: 3.8.9
NanoVna saver version : 0.3.9

NanoVna:  NanoVNA-F_V2, FW 0.3.0
(Is this an issue with the 'F' variant?)

If there is anyway I can help debugging this, let me know.
Thanks!