NanoVNA-Saver/nanovna-saver #396: NanoVNASaver 0.3.8 crashes connecting to "nanoVNA v2 +4" device "Timeout reading version registers"

zeirix, 2021-05-20:
**Describe the bug**
Nanovna-saver crashes, after clicking the "Connect to device" button.
The bug exists on both, a Windows 10 install and a Linux system, see below.
It makes no difference, if it is connected directly or via a USB-Hub.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect nanoVNA v2 +4 to computer, turn it on
2. Start NanoVNASaver, running $ python nanovna-saver.py
3. Select the correct tty Port, in this case /dev/ttyACM0 (S-A-A-2)
4. Click on [ Connect to device ]
5. NanoVNASaver crashes, outputting the error `NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - Timeout reading version registers`

**Expected behavior**
The expected behaviour would be, that NanoVNASaver connects to the device.

**Screenshots**
  ```
Settings: /home/janis/.config/NanoVNASaver/NanoVNASaver.ini
2021-05-20 13:00:45,664 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - Timeout reading version registers
Traceback (most recent call last):
File "/home/janis/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 516, in serialButtonClick
    self.connect_device()
  File "/home/janis/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 538, in connect_device
    self.vna = get_VNA(self.interface)
  File "/home/janis/nanovna-saver/NanoVNASaver/Hardware/Hardware.py", line 88, in get_VNA
    return NanoVNA_V2(iface)
  File "/home/janis/nanovna-saver/NanoVNASaver/Hardware/NanoVNA_V2.py", line 67, in __init__
    super().__init__(iface)
  File "/home/janis/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 62, in __init__
    self.read_features()
  File "/home/janis/nanovna-saver/NanoVNASaver/Hardware/NanoVNA_V2.py", line 97, in read_features
    if self.version <= Version("1.0.1"):
  File "/home/janis/nanovna-saver/NanoVNASaver/Version.py", line 60, in __ge__
    return self > other or self == other
  File "/home/janis/nanovna-saver/NanoVNASaver/Version.py", line 48, in __gt__
    l, r = self.data, other.data
AttributeError: 'NoneType' object has no attribute 'data'
```

**Desktop (please complete the following information):**
1st host:
 - OS: Linux 5.4.118-1-MANJARO
 - Python version: 3.9.5
 - NanoVNASaver version: 0.3.8

2nd host:
 - OS: Windows10 20H2
 - Python version: 3.8.7
 - NanoVNASaver version: 0.3.8

Thank You!

zeirix, 2021-05-21:
**Add:**
Saving the VNA device configuration for the first time via touchscreen, solved the problem:
Main menu > CONFIG > [ SAVE ]

Anyway, this is a point, that should be fixed.

brian-swiftlabs, 2021-06-28:
Seeing the same issue on 0.3.8 and 0.3.9 running on Windows 10 21H1.  Rescan displays correct COM port, but app crashes after 'connect to device' is pressed.
[image]

zarath, 2021-06-28:
Which firmware is on the NanoVNA V2? It crashes in Version check

closed 2022-01-06 by zarath