NanoVNA-Saver/nanovna-saver #398: Wrong axis

vreijs, 2021-06-10:
**Describe the bug**
If I use the through cable between S11 and S21 (port0 to port1). The power on Port1 is 0dB (when looking on my NanoVNA (as expected)). I see also this ~0dB  at 'S21 Gai'n value in NanaVNA Saver 'Show Data'. But in the graph it looks as if the curve S21 LogMag power is -47dB.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect through cable on NanoVNA (after calibrating, not necessary I think)
2. Do a sweep on NanoVNA Saver and look at 'Show Data' (S21 Gain. which is around 0.006dB) and Display chart: S11 & S21 LogMag (around  -47dB)

**Expected behavior**
I would expect that the S21 Gain would also be graphed closer to zero (aka the axis should be adjusted?)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - NanoVNA-Saver version:  0.3.8

**Additional context**
None

rjordans, 2021-06-15:
I see two lines in the graph, green for S21 at 0 and red for S11 at -47dB.  Maybe not the best color-scheme if you have trouble distinguishing those?

vreijs, 2021-06-15:
Thanks. I now see it, I swapped the colored *(niot due to my sight, but just because I interpret it wrongly). 
Furthermore I was trying in Display-setup.. . ->Setting to get S11in dB, but it only gives |s11| as 0.004 (which is indeed -47dB).
Thanks for your help!

rjordans, 2021-06-15:
Great, good to hear that it's 'fixed' then!

closed 2021-06-18 by zarath