NanoVNA-Saver/nanovna-saver #403: 0.39 Crashes Upon Application Startup

hammyMarc, 2021-06-21:
I get terminal window with a fast rolling dialogue of what appears to be a startup routine. That terminal window closes. Then nothing.

Is there a way I can launch it from the CLI and capture debugging information for you guys?

73, W6MGV Marc

whalphen, 2021-06-21:
I'm having the same problem.  Using Windows 10.  This is what comes up in the terminal window:

Traceback (most recent call last):
  File "nanovna-saver.py", line 21, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\__main__.py", line 36, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\NanoVNASaver.py", line 28, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Windows\__init__.py", line 2, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Windows\AnalysisWindow.py", line 23, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Analysis\__init__.py", line 1, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver\Analysis\Analysis.py", line 22, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\signal\__init__.py", line 302, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\signal\filter_design.py", line 16, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\optimize\__init__.py", line 421, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\optimize\_shgo.py", line 9, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\spatial\__init__.py", line 107, in <module>
  File "c:\hostedtoolcache\windows\python\3.8.10\x86\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 493, in exec_module
  File "scipy\spatial\transform\__init__.py", line 19, in <module>
  File "rotation.pyx", line 5, in init scipy.spatial.transform.rotation
ModuleNotFoundError: No module named 'scipy.spatial.transform._rotation_groups'
[1464] Failed to execute script nanovna-saver

akinad3, 2021-06-21:
I remember seeing this when I built an executable from a development branch. I think it was a scipy version.

zarath, 2021-06-21:
x64 or 32bit windows?

whalphen, 2021-06-21:
I'm using x64 windows.

hammyMarc, 2021-06-21:
Win10 x64

zarath, 2021-06-21:
Can one try the assets from
https://github.com/NanoVNA-Saver/nanovna-saver/actions/runs/957794231

it's in fact vß.3.9 but build with a newer pyinstaller version

whalphen, 2021-06-21:
Success!  This version works.  There is another issue, though.  The nanoVNA-F  grabs a COM port that is already assigned to another application.  It crashes unless I disable the use of the COM port by the other application.

zarath, 2021-06-21:
what kind of device is on the other port?

hammyMarc, 2021-06-21:
There were no other external devices on the USB bus, at the time.

whalphen, 2021-06-21:
I was able to manually reassign the COM port for the nanoVNA-F and it now coexists peacefully with the other application.  The other application is Eltima Virtual Serial Port Driver.  I've occasionally seen other applications that try to take the ports assigned to this driver.  I suppose there is a way to avoid it in software, but perhaps some applications don't follow the safest approach to grabbing a COM port.

Anyway, I manually reassigned the port used by the nanoVNA-F by right clicking on the port listing in Device Manager and selecting Properties, Advanced..., COM Port Number.

zarath, 2021-06-21:
This USB-Serial port drivers are sometimes somewhat crazy.

hammyMarc, 2021-06-21:
New build works fine, with the exact configuration that failed yesterday. Swept my Diamond dual band to make sure. Thanks for the fix! (wish my SWR were a smidge better LOL but VNASaver is working like a champ, now).

zarath, 2021-06-21:
Replaced Windows binaries by builds made with pyinstaller 4.3

closed 2021-06-21 by zarath

hammyMarc, 2021-06-21:
In the future, hit me up to QA your release candidates. Happy to help the effort. It's the least I can do. Great software and thank you for your hard work! -W6MGV

whalphen, 2021-06-21:
Thanks for your help!

K8VFO