NanoVNA-Saver/nanovna-saver #405: NanoVNA-Saver 0.3.10-pre (0.3.9) with NanoVNA-H4 Datapoints setting

PA3CCE, 2021-06-24:
**Describe the bug**
=============
Datapoints setting for NanoVNA-H4  Firmware V 1.0.65 problematic.

**To Reproduce**
Steps to reproduce the behavior:
=====================
1. Go to "Manage"en select ""Datapoints"
    One can select 11, 51, 101, 201 and 401 points. (default is 101 points).
    a)  11 points is NO more supported by the device.
    b)  when 401 points is selected an Error-screen pops up when "Sweep"' is started :
    
   "Error during sweep
    Stopped
    list index out of tange"

**Expected behavior**
   Is obvious
 
**Desktop (please complete the following information):**
 - OS: Windows 10:
 - NanoVNA-Saver version: 0.3.10-pre (0.3.9)
 - NanoVNA-H4 Firmware by DiSllord version 1.0.65

 - Note: 
 - When a NanoVNA V2 device  with it's latest firmware by DiSlord is used all number of datapoints provided work excellent !

PA3CCE, 2022-10-10:
I checked this in the latest version (5;3) and the problem still is present !

sfabris, 2023-02-07:
Still present in 5.4

closed 2023-06-10 by github-actions[bot]