NanoVNA-Saver/nanovna-saver #411: After building it, NanoVNA-Saver 0.3.10rc2 not running

PA3CCE, 2021-06-28:
**Describe the bug**

After "Successfully installed NanoVNA-Saver-0.3.10rc02"  in the command-window and running nanovnasaver.exe I finally get :

NanoVNASaver.py , line27 , in <module> from .Windows Import ( ModuleNotFoundError: No module named 'NanoVNASaver.windows'

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Python version: 3.9
 - NanoVNA-Saver version: NA

**Additional context**
NanoVNASaver 03.10rc01 is running well after building.

zarath, 2021-06-28:
0.3.10rc2???

zarath, 2021-06-29:
Ok. I see setuptools makes an rc out of pre in its version evaluation ;-)

PA3CCE, 2021-06-29:
Ok, downloaded the latest sources, but now I get a similar message "no module named  NaanobnaSaver.Hardware.TinySA.

Sorry,
Jos

zarath, 2021-06-29:
Strange...

PA3CCE, 2021-06-30:
This morning I downloaded the latest sources and **rc4** runs fine !!
Thank you,
Jos

commit 14858ab7 (2021-06-29): modified setup.cfg. should fix #411
closed 2021-07-01 by zarath