NanoVNA-Saver/nanovna-saver #418: No module named 'scipy.spatial.transform._rotation_groups

sprakashb, 2021-07-17:
Getting "No module named 'scipy.spatial.transform._rotation_groups " error in ver 0.3.9 for linux. 
Tried to reinstall scipy but same error. 
Ver 0.3.8 working well.

qbit-dev, 2021-07-19:
The same here. Ubuntu 20.04.2

Yume303, 2021-07-28:
Remove the python3 serial module and leave only pyserial intalled, that solved the above error for me.

qbit-dev, 2021-07-28:
@Yume303 Didn't work for me - I'm getting the same error.

zerog2k, 2021-07-29:
also same on Mac OSX Mojave with downloaded 0.3.9 release
I have scipy installed on my py3 install, however this is some self-executable/encapsulated py, so maybe it's missing from there?

```
$ ./nanovna-saver
Traceback (most recent call last):
  File "nanovna-saver.py", line 21, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/__main__.py", line 36, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/NanoVNASaver.py", line 28, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/Windows/__init__.py", line 2, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/Windows/AnalysisWindow.py", line 23, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/Analysis/__init__.py", line 1, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "NanoVNASaver/Analysis/Analysis.py", line 22, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/signal/__init__.py", line 302, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/signal/filter_design.py", line 16, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/optimize/__init__.py", line 421, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/optimize/_shgo.py", line 9, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/spatial/__init__.py", line 107, in <module>
  File "/Users/runner/hostedtoolcache/Python/3.8.10/x64/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 493, in exec_module
  File "scipy/spatial/transform/__init__.py", line 19, in <module>
  File "rotation.pyx", line 5, in init scipy.spatial.transform.rotation
ModuleNotFoundError: No module named 'scipy.spatial.transform._rotation_groups'
```

as others reports, the 0.3.8 binary works fine

Yume303, 2021-07-29:
Ohh sorry, compilled the 0.3.9 from source for myself, after that removed the installed serial module, leaving the pyserial "alone, and it worked on 21.04.

errolt, 2022-01-01:
> after that removed the installed serial module,
I can't remove installed modules as that would break other programs.