NanoVNA-Saver/nanovna-saver #420: 0.3.9 segmentation fault on start-up on Arch Linux

LeightonMan, 2021-07-23:
**Describe the bug**
On running the latest downloaded version, nanovna-saver crashes with the following output:
NanoVNASaver 0.3.8

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Segmentation fault (core dumped)

**To Reproduce**
Steps to reproduce the behavior:
1.download binary, unzip, and make executable with chmod u+x nano-vnasaver
2. run in terminal with ./nanovna-saver

**Desktop (please complete the following information):**
 - OS: Arch Linux x86_64 kernel version 5.13.4
 - Python version: 3.9.6
 - NanoVNA-Saver version: 3.9 

**Additional context**
Also happens with 3.8 which ran correctly on a different machine also using Arch Linux

goblenc, 2022-02-21:
I have the same issue on Debian:

./nanovna-saver 
NanoVNASaver 0.3.10

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020, 2021 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Segmentation fault

Debian 11
Python 3.9
NanoVNA-Saver 3.10

Jojeee-xyz, 2022-03-30:
Also having this issue on Manjaro (Arch):

./nanovna-saver
NanoVNASaver 0.3.10

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020, 2021 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Segmentation fault (core dumped)

To reproduce download and extract NanoVNA-Saver
chmod +x
execute the program.

Manjaro Linux 5.10
Python 3.10.2
NanoVNA-Saver 0.3.10

Jojeee-xyz, 2022-03-30:
Reproduced fault with versions of NanoVNA-Saver 0.3.4 onwards

zarath, 2022-10-07:
several crashes smelled like python 3.10