NanoVNA-Saver/nanovna-saver #434: nanovna-save crashes when connecting to Aursinc nanovna SAA-2N

pburns001, 2021-09-22:
[image]

pburns001, 2021-09-22:
If I already have it in usb mode by first connecting to VNA_QT and terminating vna_qt, it connect without error.

JamesLPearson, 2021-12-28:
I have the same issue. 

> If I already have it in usb mode by first connecting to VNA_QT and terminating vna_qt, it connect without error.
Thank you for that workaround - that worked for me too. Opened VNA_QT, selected the COM port and connected. That put the NanoVNA into 'USB Mode', which allowed NanoVNA_Saver to work. 

Seems like NanoVNA _Saver is not making the device enter USB Mode for some reason.

closed 2022-11-10 by github-actions[bot]