NanoVNA-Saver/nanovna-saver #435: an unhandled win32 exception occured in nano-saver.exe [4304]

iutech-france, 2021-09-28:
**Describe the bug**
On launching nanosaver.exe (both version 0.3.9 and 0.3.8) on Windows7 I get the message "an unhandled win32 exception occured in nano-saver.exe [4304]" and then I have only two options, "debug" (which doesn't seem to produce anything) and "close the program".

**To Reproduce**
Steps to reproduce the behavior:
Click on nanovna-saver.exe

**Expected behavior**
That nanoVNA saver launches without crashing just afterwards.

.
**Desktop (please complete the following information):**
 - Windows 7
 - Python version: how to check ? Not sure there is one installed
 - NanoVNA-Saver version: [both 0.3.9 and 0.3.8]

iutech-france, 2021-09-29:
Trying on a Windows 10 VM the problem doesn't appear; but on physical machines the crash appeared systematically on the two Windows 10 computers I tried nanovna-saver on.

iutech-france, 2021-10-07:
Also it offers to debug the crash for me, but the default debugger is Microsoft Visual Studio that refuses to work due to a license problem.
Which other debuggers could be used ?
Is there a way to find log files somewhere that could explain why it crashed ?

zarath, 2022-01-04:
No resources for fixing build on outdated/obsolete windows versions.
Perhaps newest release will work?

closed 2022-01-04 by zarath