NanoVNA-Saver/nanovna-saver #442: NanoVNA V2 H4 Support - Development Questions

curiousmuch, 2021-11-23:
Hello! 

I'm working with the NanoVNA V2 H4. I'm creating a scripting interface for use with automated coupon measurements. I've already based most of the work on the current NanoVNA_V2 class. I'd like to push this back to NanoVNA-Saver in the near future. 

(1) Would it be acceptable to use the board revision to link to my new class NanoVNA_V2_H4 using the get_VNA function? 
(2) Some of the methods for the class have arguments which I feel like what might be legacy i.e.("data 0", "data 1"). I'd like to optimize this do the developers have any comments? In addition, I'd like to return the data as a complex values and not a string of complex values...
(3) The USB mode for the H4 allows 65,000+ data points, but the current UI limits this to small subset. Is their UI reasons for this? I've already tested the scriptable interface to 10,000 points and it works well. 
(4) Would the developers prefer I used pep8 for methods or camelcase?

zarath, 2021-11-24:
Hello ;-)

1) of course
2) the whole worker is legacy. I too dislike that string data, but had no time to refactor and test all dependent hardware classes.
3) whole ui classes gone wild and should be refactored (much work)
4) That's a problem when using qt with python. As qt uses CamelCase try to use CamelCase in qt inherited classes but prefer snake_case if it is independent of qt.

curiousmuch, 2021-11-24:
Thanks Zarath! 
I'll have the code work working for the scripting interface, and then review the worker threads. I'll try to think of a few methods to integrate them and post back here for review before implementation. 

Also, I simply want to say thanks to you and any of the other developers working on this project.

closed 2021-11-24 by curiousmuch