NanoVNA-Saver/nanovna-saver #444: Crashes on Connect

BlackGuns, 2021-12-08:
**Describe the bug**
When connecting to the NanoVNA the app closes.

**To Reproduce**
1. Connect VNA to USB port 
2. Power on VNA
3. Click refresh on the saver software (my computer shows Com6 has a VNA attached)
4. Click Connect - app crashes

**Expected behavior**
I would expect the app to not crash.  ;-)

**Screenshots**
Unfortunately the terminal window in the background disappears when the app crashes.  If someone can tell me how to "catch" it before it closes I can snip and attach a pic.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro, Version 10.0.19042 Build 19042, x64 based PC
 - Python version: N/A
 - NanoVNA-Saver version: 0.3.9

**Additional context**
Using a Sysjoint NanoVNA-F V2 with firmware 0.3.0 and Rev. C hardware.

dmazzer, 2021-12-14:
Same problem here with exact same versions used by @BlackGuns.

This are the messages from windows console.

```
PS C:\Users\x\Downloads\NanoVNASaver.x64> .\nanovna-saver.exe
NanoVNASaver 0.3.9

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: C:/Users/dmz/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2021-12-14 10:50:28,715 - NanoVNASaver.NanoVNASaver - ERROR - Tried to open COM4 (NanoVNA) and failed: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
Traceback (most recent call last):
  File "NanoVNASaver\NanoVNASaver.py", line 522, in serialButtonClick
  File "NanoVNASaver\NanoVNASaver.py", line 542, in connect_device
  File "serial\serialutil.py", line 372, in timeout
  File "serial\serialwin32.py", line 222, in _reconfigure_port
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

The error message is thrown when the connect button is pressed.

Hope it helps.

JamesLPearson, 2021-12-28:
I'm having trouble connecting too. It was working about a year ago and nothing has changed on the NanoVNA. I even tried running the previous version of NanoVNA-Saver (0.3.8), which was previously working and the issue remains. Something else must be causing the trouble.

Here is the contents of the command window before it closes:

```
NanoVNASaver 0.3.9

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: C:/Users/james/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
2021-12-28 16:29:10,957 - NanoVNASaver.Hardware.NanoVNA_V2 - ERROR - Timeout reading version registers
Traceback (most recent call last):
  File "NanoVNASaver\NanoVNASaver.py", line 522, in serialButtonClick
  File "NanoVNASaver\NanoVNASaver.py", line 545, in connect_device
  File "NanoVNASaver\Hardware\Hardware.py", line 88, in get_VNA
  File "NanoVNASaver\Hardware\NanoVNA_V2.py", line 76, in __init__
  File "NanoVNASaver\Hardware\VNA.py", line 66, in __init__
  File "NanoVNASaver\Hardware\NanoVNA_V2.py", line 107, in read_features
  File "NanoVNASaver\Version.py", line 63, in __le__
  File "NanoVNASaver\Version.py", line 57, in __lt__
  File "NanoVNASaver\Version.py", line 57, in __lt__
  File "NanoVNASaver\Version.py", line 57, in __lt__
  [Previous line repeated 490 more times]
RecursionError: maximum recursion depth exceeded

```

silbe, 2022-03-20:
@JamesLPearson You're seeing a different problem than the one reported by @BlackGuns. In your case, opening the port works fine but communication with the NanoVNA fails. The infinite recursion during version comparison has been fixed in the `testing` branch (see #475) but that's only a symptom itself, not the cause.

JamesLPearson, 2022-03-20:
> @JamesLPearson You're seeing a different problem than the one reported by @BlackGuns. In your case, opening the port works fine but communication with the NanoVNA fails. The infinite recursion [...]

Ah yes, I see the difference. Thank you very much for letting me know and for pointing me towards the branch that resolves this.