NanoVNA-Saver/nanovna-saver #445: Wrong units shown for admittance values

retiredprogrammer54, 2021-12-13:
The marker data for admittance use ohms as a unit. This should be S (for Siemens) or mho or Ω-1 (-1 as a power)

[image]
.

rjordans, 2021-12-14:
Actually, it looks like the calculation is not showing the admittance but the parallel impedance so either the calculation or the labeling of the value needs to change

rjordans, 2021-12-14:
I believe the above pull request should fix the issue, can you check if it works for you?

DiSlord, 2021-12-21:
Saver show correct values in Om
See https://en.wikipedia.org/wiki/Admittance
G = R / (R*R + X*X)
B = - X / (R*R + X*X)
This result in Siemens as you write. Siemns = 1 / Om

But NanoVNA Saver use different formula:
Rp = (R*R + X*X) / R !!! this result in Om
Xp = (R*R + X*X)  / X !!! this result in Om but lost sign??

rjordans, 2021-12-21:
Well, as the link to Wikipedia shows, admittance is defined as 1/Z and has the unit Siemens.  If you change the calculation you get something different which is not what I, and apparently others, would expect when you label it admittance.

The formula that NanoVNA Saver currently uses could be called something like parallel impedance model or so, but as long as it's called admittance by NanoVNA Saver it will cause confusion.

My change in #446 switches the calculation of admittance to provide the result in Siemens.  I'd suggest to either do it that way or to re-label the admittance field, as it's currently presented, in a way that avoids this confusion.

DiSlord, 2021-12-21:
While add admittance in NanoVNA firmware i add additional markers:
  [MS_GB]   = "G + jB", // in Siemens 
  [MS_GLC]  = "G + L/C", // G in Siemens 
  [MS_RpXp] = "Rp + jXp", // In Om
  [MS_RpLC] = "Rp + L/C", // Rp in Om

Most question, need change sign for Xp?
B = -X / (R * R + X * X)

rjordans, 2021-12-21:
My first guess would be that you may have a j^2 in there that gives you the -1 factor.

Quick check: for a capacitor I'd expect some negative value for X, and a positive value for B so that would also match the formula

PR #446 (2022-01-04): Change calculation of admittance value and unit in marker plotting code
Calculates the admittance as 1/Z, checks for Z=0.

Unit is "S" for Siemens as Omega^-1 was running into space issues in the currently available formatting of the marker window

Fixes #445
closed 2022-11-10 by github-actions[bot]