NanoVNA-Saver/nanovna-saver #449: Issue in apply calibration standarts Open offset delay

DiSlord, 2021-12-27:
**Describe the bug**
If set C0, C1, C2, C3 values to 0
then Offset delay not apply

**To Reproduce**
Made calibration, measure open, see point at right

If set C0, C1, C2, C3 values to 0
set Offset delay 100
measure open, see again point at right, but this not correct, must be arc

Set C0 to 1
measure open, see correct result as arc
Set offset to 0, see point also corect result

Problem in this part
    def gamma_open(self, freq: int) -> complex:
        g = Calibration.IDEAL_OPEN
        if not self.useIdealOpen:
            logger.debug("Using open calibration set values.")
            divisor = (2 * math.pi * freq * (
                self.openC0 + self.openC1 * freq +
                self.openC2 * freq**2 + self.openC3 * freq**3))
            if divisor != 0:

if set C0, C1, C2, C3 values to 0 divisor = 0

but in calibration apply divisor can be 0

DiSlord, 2021-12-29:
Solution for prevent division on 0:
Zop = complex(0, -1) / divisor
Use Zop` = 1 / Zop
In this case
Zop` = complex(0, 1) * (2 * math.pi * freq * ( ......

Then you get:
g =((Zop / 50 - 1) / (Zop / 50 + 1)) =  ((1 - 50 * Zop`) / (1 + 50 * Zop`))

For open this must written by

`    def gamma_open(self, freq: int) -> complex:
        g = Calibration.IDEAL_OPEN
        if not self.useIdealOpen:
            logger.debug("Using open calibration set values.")
            Zop = complex(0, 1) * (2 * math.pi * freq * (
                self.openC0 + self.openC1 * freq +
                self.openC2 * freq**2 + self.openC3 * freq**3))
            g = ((1 - 50 * Zop) / (1 + 50 * Zop)) * cmath.exp(
                complex(0, 1) * 2 * math.pi *
                2 * freq * self.openLength * -1)
        return g
`

DiSlord, 2021-12-29:
Also need remove warning in CalibrationSettings.py
                if self.app.calibration.openC0 == 0:
                    raise ValueError("C0 cannot be 0.")

DiSlord, 2022-01-03:
Add fixes and more in
https://github.com/NanoVNA-Saver/nanovna-saver/pull/450

zarath, 2022-01-04:
Thanks for your contribution!

closed 2022-01-04 by zarath