NanoVNA-Saver/nanovna-saver #453: NanoVna Saver crashes  after 1port calibration

ok1vaw, 2022-01-04:
**Describe the bug**
A clear and concise description of what the bug is.

**When performing a simple 1port SOL calibration in the range of 1MHz to 30MHz with 101 points using the NAnovnaV2s NanovnaSaver crashes.**

**To Reproduce**
Steps to reproduce the behavior:
1. Start NanoVna saver with NanoVNa V2
2.  Set range of 1MHz to 30MHz with 101 points 
3.  Use calibration assistant, calibrate with SOL standards
4.  When applying calibration, the Saver crashes

**Expected behavior**
A clear and concise description of what you expected to happen.
The calibration should be applied to measurement, there should be no crash.

**Screenshots**
Contents of the debug log:
2021-12-29 21:01:36,873 - NanoVNASaver.Calibration - DEBUG - Calibration correctly calculated.
2021-12-29 21:01:36,874 - NanoVNASaver.Windows.CalibrationSettings - DEBUG - Applying calibration to existing sweep data.
2021-12-29 21:01:36,897 - NanoVNASaver.Windows.CalibrationSettings - DEBUG - Saving and displaying corrected data.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Win10
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.3.9]

**Additional context**
Add any other context about the problem here.

zarath, 2022-01-04:
Still existing with v0.3.10 ?

zarath, 2022-01-09:
Seen a comment in a mailing list, that it seemed to be fixed

closed 2022-01-09 by zarath