NanoVNA-Saver/nanovna-saver #456: Please use more relaxed deps in the setup.cfg

yarda, 2022-01-10:
**Describe the bug**
The commit a3c9dea92fa1bdc4cea5c509575fa058ee86bfda caused problems when doing distro packaging, e.g.:
https://bugzilla.redhat.com/show_bug.cgi?id=2038547
Could you please use more relaxed deps?

**To Reproduce**
Steps to reproduce the behavior:
1. Check deps in the setup.cfg

**Expected behavior**
Relaxed versions to match most distros current packages.

**Desktop (please complete the following information):**
 - OS: Fedora 36
 - Python version: python-3.10.1
 - NanoVNA-Saver version: nanovna-saver-0.3.10

barjac, 2022-04-09:
Similar issue trying to install our nanovna-saver-0.4.0 package for Mageia9 which uses python3-numpy-1.22.0.
I have patched setup.cfg and so far I see no regressions:
-    numpy>=1.21.1,<1.22.0
+    numpy>=1.21.1
These over restrictive version limits cause real headaches for packagers.

zarath, 2022-04-09:
I have had regressions with numpy 1.22. But it could be that they were related by python-installer. Therefore the spec to a version below 1.22.

But I'll take a look if 1.22 works.

I already have different dependency settings for binary vs source installation.

barjac, 2022-04-10:
Do you recall the regressions that you saw?
I can test any specific functions if I know what to look for.
I do use this myself with the original nanovna.

commit 6aa7aaa0 (2022-05-14): Feature/linting 220402 (#499)

* added .flatpak-builder to .gitingnore
* GroupDelay Chart simplified
* allow numpy > 1.21 fixes #456
* Added flatpak manifest
closed 2022-05-14 by zarath