NanoVNA-Saver/nanovna-saver #457: Aborted (core dumped) on Fedora 35 (Python 3.10.1), NanoVNA H4, DisLord 1.0.45

galileo-pkm, 2022-01-10:
Crashes with various messages.
Most of the time it does not detect the device properties, on connect it
either crashes or shows "list index out of range". Sometimes it crashes 
when attempting to change device setting (BW, datapoints).
This was tested with master and Development branch. More or less
the same behaviour. 

A selection of errors:

```

[galileo@sol nanovna-saver (Development)]$ python nanovna-saver.py 
NanoVNASaver 0.3.11-pre

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

2022-01-10 22:37:40,477 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 79, in run
    self._run()
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 118, in _run
    freq, values11, values21 = self.readAveragedSegment(
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 233, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 265, in readSegment
    values11 = self.readData("data 0")
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 280, in readData
    tmpdata = self.app.vna.readValues(data)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/NanoVNA.py", line 144, in readValues
    f"{data[0]} {data[1]}",
IndexError: list index out of range
2022-01-10 22:38:03,426 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 79, in run
    self._run()
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 118, in _run
    freq, values11, values21 = self.readAveragedSegment(
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 233, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 265, in readSegment
    values11 = self.readData("data 0")
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 280, in readData
    tmpdata = self.app.vna.readValues(data)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/NanoVNA.py", line 145, in readValues
    f"{data[2]} {data[3]}"))
IndexError: list index out of range
2022-01-10 22:38:15,225 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 79, in run
    self._run()
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 118, in _run
    freq, values11, values21 = self.readAveragedSegment(
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 233, in readAveragedSegment
    freq, tmp11, tmp21 = self.readSegment(start, stop)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 265, in readSegment
    values11 = self.readData("data 0")
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 280, in readData
    tmpdata = self.app.vna.readValues(data)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/NanoVNA.py", line 144, in readValues
    f"{data[0]} {data[1]}",
IndexError: list index out of range
2022-01-10 22:38:31,230 - NanoVNASaver.SweepWorker - ERROR - list index out of range
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/SweepWorker.py", line 79, in run
[... 62 lines cut ...]
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

2022-01-10 22:17:56,941 - NanoVNASaver.Controls.SerialControl - ERROR - Unable to connect to VNA: too many retries
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Controls/SerialControl.py", line 73, in serialButtonClick
    self.connect_device()
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Controls/SerialControl.py", line 104, in connect_device
    frequencies = self.app.vna.readFrequencies()
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 148, in readFrequencies
    return [int(f) for f in self.readValues("frequencies")]
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 190, in readValues
    result = list(self.exec_command(value))
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/VNA.py", line 92, in exec_command
    drain_serial(self.serial)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Hardware/Serial.py", line 33, in drain_serial
    cnt = len(serial_port.read(128))
  File "/usr/lib/python3.10/site-packages/serial/serialposix.py", line 478, in read
    raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open
Aborted (core dumped)

```

galileo-pkm, 2022-01-10:
This is probably related to changes in Python, version 0.3.7 crashes with
this error:

```

[galileo@sol nanovna-saver-0.3.7]$ python nanovna-saver.py 
NanoVNASaver 0.3.7

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: /home/galileo/.config/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.7/NanoVNASaver/Charts/Smith.py", line 46, in paintEvent
    self.drawValues(qp)
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.7/NanoVNASaver/Charts/Smith.py", line 165, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m)+1)
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.7/NanoVNASaver/Charts/Chart.py", line 315, in drawMarker
    qp.drawText(number_x, number_y, str(number))
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)

```
Same issue with 0.3.8

```
[galileo@sol nanovna-saver-0.3.8]$ python nanovna-saver.py
NanoVNASaver 0.3.8

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Settings: /home/galileo/.config/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.8/nanovna-saver-0.3.8/NanoVNASaver/Charts/Smith.py", line 46, in paintEvent
    self.drawValues(qp)
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.8/nanovna-saver-0.3.8/NanoVNASaver/Charts/Smith.py", line 165, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m)+1)
  File "/home/galileo/tmp/nanovna/nanovna-saver-0.3.8/nanovna-saver-0.3.8/NanoVNASaver/Charts/Chart.py", line 315, in drawMarker
    qp.drawText(number_x, number_y, str(number))
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)

```

galileo-pkm, 2022-01-11:
Saver also crashes when loading s2p files. Tested with develop and master branch.
Interesting bit is that it does not crash when loading the file as reference, only as
a sweep. 

Similar errors as before, s2p file that causes the crash was saved using 0.3.9 version (latest does not work
under win7 due to python issues) and is attached. 
[H4-1.1.s2p.gz](https://github.com/NanoVNA-Saver/nanovna-saver/files/7842759/H4-1.1.s2p.gz)

```
[galileo@sol nanovna-saver (master)]$ python nanovna-saver.py 
NanoVNASaver 0.3.10

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020, 2021 NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Qt: Session management error: networkIdsList argument is NULL
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Smith.py", line 47, in paintEvent
    self.drawValues(qp)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Smith.py", line 166, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m)+1)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Chart.py", line 277, in drawMarker
    cmarker.draw(x, y, color, str(number))
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Chart.py", line 100, in draw
    self.qp.drawText(x - text_width // 2, y - 3 - offset, text)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)
[galileo@sol nanovna-saver (master)]$ 
[galileo@sol nanovna-saver (master)]$ 
[galileo@sol nanovna-saver (master)]$ git checkout Development 
Switched to branch 'Development'
[galileo@sol nanovna-saver (Development)]$ python nanovna-saver.py 
NanoVNASaver 0.3.11-pre

Copyright (C) 2019, 2020 Rune B. Broberg
Copyright (C) 2020ff NanoVNA-Saver Authors

This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/NanoVNA-Saver/nanovna-saver for further details.

Qt: Session management error: networkIdsList argument is NULL
Traceback (most recent call last):
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Smith.py", line 47, in paintEvent
    self.drawValues(qp)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Smith.py", line 166, in drawValues
    self.drawMarker(x, y, qp, m.color, self.markers.index(m)+1)
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Chart.py", line 277, in drawMarker
    cmarker.draw(x, y, color, str(number))
  File "/home/galileo/tmp/nanovna/saver-git/galileo/nanovna-saver/NanoVNASaver/Charts/Chart.py", line 100, in draw
    self.qp.drawText(x - text_width // 2, y - 3 - offset, text)
TypeError: arguments did not match any overloaded call:
  drawText(self, Union[QPointF, QPoint], str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRect, int, str): argument 1 has unexpected type 'int'
  drawText(self, QRectF, str, option: QTextOption = QTextOption()): argument 1 has unexpected type 'int'
  drawText(self, QPoint, str): argument 1 has unexpected type 'int'
  drawText(self, int, int, int, int, int, str): argument 2 has unexpected type 'float'
  drawText(self, int, int, str): argument 2 has unexpected type 'float'
Aborted (core dumped)

```

galileo-pkm, 2022-01-11:
Traced the issue down to Python version, running saver from a virtual environment with Python 3.9 resolves all observed issues.

zarath, 2022-09-18:
v0.5.3 and later should be python 3.10 compatible

closed 2022-09-18 by zarath