NanoVNA-Saver/nanovna-saver #459: No calibration with nanovna-f v2 and Fmin=50kHz, Fmax = 3F, 5 segments

roland-bruns, 2022-01-11:
Using 50kHz as minimum and 3GHz as the maximum frequency for a calibration with five segments leads to the following Error:
Terminal:  **NanoVNASaver.Calibration - ERROR - Division error - did you use the same measurement for two of short, open and load?**
GUI : **Error applying calibration - Two of short, open and load returned the same values at frequency **3000000128Hz****
Using 2.9GHz for the maximum frequency runs without problems. Seems to be a rounding problem, the shown frequency is > 3GHz

[image]

- Linux Mint 20.3 
- Python 3.8
- NanoVNA-Saver version: 0.3.10
- NANOVNA-F V2, Firmware v0.3.0 / v0.3.2 (Firmware update makes no difference)

roland-bruns, 2022-01-13:
The problem seems to be in Sweep.py, line 85:
_return round(self.span / (self.points  * self.segments - 1))_ is not o.k. 
It should be:
 _return round(self.span / (self.points  * self.segments ) - 1)_  (move '-1' behind the first bracket)
But even this is not the best solution, it would be better to use:
_return math.floor(self.span / (self.points  * self.segments))_

byteptr, 2022-09-03:
I have NanoVna V2 Plus5 Pro (the original as recommends nanofre site). 
I tried on windows 10 and I have exactly the same Error but at 200 MHz even if I set an arbitrary range  such as Sweep control to Start: 1MHz, Stop 4.4GHz. and manual calibration. (I don't know which firmware version is used here).
[image]
So we can infer that is not a platform dependant and this agrees with the bunch of code code suggeted before.
Bun there are other things to check, if I try to use calibration assistant with aforementioned range  or another one set by me, when I click to "Apply" button, the software changes unexpectedly Sweep control to default settings (Start 200Mhz, Stop 300MHz), a little behavior which can turn the user experience very annoying, and even worse It can becomes totally impractical for a potential amazing software solution which have a nice look and lot of options.
[image]
I have done some search on the net, but I nothing related found. I will recheck and list all of the steps to docment and reproduce the issue, and try to open a new one here.
Best regards.

closed 2023-01-02 by github-actions[bot]