NanoVNA-Saver/nanovna-saver #460: NanoVNA-Saver version 3.10 fails in startup

PA3CCE, 2022-01-12:
When I start Nanovna-Saver 3.10 and connect it to the device  (NanoVNA-H4 on COM10)  it performs a sweep and shows : 

Error during sweep 
stopped
Failed reading data 1-10 times
Data outside expected valid range or in an unexpected format
you can disable datavalidationon the device screen

Windsow's commandscreen shows :

NanoVNA Saver.sweepworker warning.
Got a non plausible data value  (9.651182144 2.679436928)  <---> 10 timess

NanoVNASaver.py line  79, in run
------- " "--------- ine118 in_run
------- " ---------- line 233 read average segment
------- " ---------- line 286 read segment
------- " ---------- l;ine 318 read data 

Operating system : Windows 11
Firmware :  DiSlord's version 1.1.01 (dec 25 2021)
Data validation NOT activated.
The error message above  is repeated  each time I try to measure or calibrate.

gaionim, 2022-01-12:
Is nanovna calibrated ? 
If not, can You try calibrate it ( standalone) and after connect to PC ?

PA3CCE, 2022-01-15:
No,  I newly installed the updated version 3.10, started it and tried to connect it to the NanoVNA-H4, then the warning popped up.
and it keeps popping up !
I did not have this with with the previous version.

gaionim, 2022-01-16:
So NanoVNA-H4 is not calibrated.
Can You try to calibrate BEFORE connect and see if the problem is the same ?

PA3CCE, 2022-01-16:
OK, this is what I did :

1) Downloaded the sources with the very latest update from the development in Gihub.
2)  Made the scrpt-file with PiIP3 and ran it.   (result : NanoVNA-Saver v 3.11 pre),    I did NOT connect to the device !
3) Tried to do a calibration   ( result :  a message that I shall have to connect to the device first).
3) Conneected to the device   ( result : same error as described earlier).

gaionim, 2022-01-16:
I'm sorry for my english, but i mean to calibrate nanovna bi itself like this https://youtu.be/EUPzbkDl4oU ( only the first link I've found )

PA3CCE, 2022-01-18:
OK, no problem, yes my NanoVNA-H4 has been calibrated.

gaionim, 2022-01-19:
Ok, I can't help, I'm missing something.
For me data is s11 ( or s21 ) and must be [-1,1] . 
I don't understand. 
The check seems in SweepWorker.py
```
                    if self.app.vna.validateInput and (
                            abs(float(a)) > 9.5 or
                            abs(float(b)) > 9.5):
```

1) You say validateInput is false, so the check  must not run.

2) Seems that valid data is [-9.5,9.5]   instead of [-1,1]

for me, valid data is        
```
 -1 <= abs(complex(float(a),float(b)) <=1
```
BUT I MAY BE WRONG !!!

3) You say that nanovna is calibrated, so the value sould be right

I leave the problem to those who are more experienced than me ;-)

gaionim, 2022-01-19:
PS
For me, old nanovna on linux, self.app.vna.validateInput is 
False   by default
0 if I check and uncheck validation flag
but in both cases do not engage validation.

PA3CCE, 2022-01-25:
OK Galonim i understand you can`t help me , anyhow many thanks for trying .
I will check some more myself and  in the mean time I hope there is someone who can fix my problem.

Best regards.

PA3CCE, 2022-01-27:
Hi Galonim,
I did some more investigations and found the solution :

!)   Before I connected to the device I opened te window where you can set the datavalidation, it was NOT SET !!!
2)  I dd SET it and tried to connect, I got the error message that I should unset the validation, so Ii UNSET it !
    Now I did SET it and tried to connect, I got the error message that I should unset the validation, so Ii did UNSET it !
    Now I ccould make the connection to the device and start aa measurement !!

Conccclusion : Though the dataavalidation initially was NOT SET, the program read that it was set.
                       SETTING and UNSETTING the datavalidation cured the problem !!

                      THE SOFTWARE SHALL HAVE TO RESET THE VALIDATION DATAFIELD BEFORE THE FIRST SART   !!!!

Kindd regards,
Jos

gaionim, 2022-01-27:
I think it could be e problem on default values in different version of python or from linux and windows.
Explicit set would be is safier ;-)
I'm not skilled on QT :-(

PA3CCE, 2022-04-10:
Solved in V 0.4.o

closed 2022-04-10 by PA3CCE

jpmcneely, 2022-06-03:
I experienced this situation with NanoVNA Saver 0.4.0 using a NanoVNA-F HW 3.1 FW 1.0.1 - There may still be an issue.
Going into manage, validation was not checked.  I checked it, then closed manage, then re-opened manage and unchecked it.  After this, I was able to connect.

Here are the last few lines from the console:
2022-06-03 08:52:08,224 - NanoVNASaver.SweepWorker - WARNING - Got a non plausible data value: (9.083205224 -14.423243520)
2022-06-03 08:52:08,425 - NanoVNASaver.SweepWorker - CRITICAL - Tried and failed to read data 1 10 times. Giving up.
2022-06-03 08:52:08,425 - NanoVNASaver.SweepWorker - ERROR - Failed reading data 1 10 times.
Data outside expected valid ranges, or in an unexpected format.

You can disable data validation on thedevice settings screen.
Traceback (most recent call last):
  File "NanoVNASaver\SweepWorker.py", line 79, in run
  File "NanoVNASaver\SweepWorker.py", line 118, in _run
  File "NanoVNASaver\SweepWorker.py", line 234, in readAveragedSegment
  File "NanoVNASaver\SweepWorker.py", line 267, in readSegment
  File "NanoVNASaver\SweepWorker.py", line 311, in readData
OSError: Failed reading data 1 10 times.
Data outside expected valid ranges, or in an unexpected format.

fredericgermain, 2022-08-10:
- NanoVNA 3.1 - random cheap 2019 model from AliExpress "Guangyi0016 Store"
- Flashed with lastest NanoVNA-H-SI_20211230.bin from https://github.com/hugen79/NanoVNA-H/releases
- nanovna-saver current main bd4eeb5e638d78473c7f1480a4fd6ffc23ecadba

I did the calibration Open+Short+Load. I got that error too on first starts, Not even knowing or changing that "validate data" in the app.

It worked by removing data validation setting in the app.

I went to dig further why I got that. It appeared only data from CH1. So after thinking a lot, I did calibration for CH1, ISOLN (plugging 50ohm on CH1?), and THRU (shorting CH1 and CH2).
After that, it worked ok even when activating data validation.

As most tutorials on internet sats to skip ISOLN+THRU, it would be advisable to skip that validation part for CH1 entirely 

Actually data validation on connection, and having some smart logic in the app to check data validation after some calibration settings are set would be advisable.

Probably a lot of people get tricked by this.