NanoVNA-Saver/nanovna-saver #464: TypeError: setValue(self, int): argument 1 has unexpected type 'float'

Werzi2001, 2022-01-15:
**Describe the bug**
NanoVNA-Saver crashes with the following error:
```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/NanoVNASaver.py", line 496, in dataUpdated
    self.sweep_control.progress_bar.setValue(self.worker.percentage)
TypeError: setValue(self, int): argument 1 has unexpected type 'float'
```

**To Reproduce**
Cannot be reproduced 100% but happens about 50% for me doing that:
1. Perform calibration (100k-30M for me)
2. Adding VSWR
3. Click on 'Sweep'
4. NanoVNA-Saver crashes with error

**Expected behavior**
No crash :-)

**Desktop (please complete the following information):**
 - OS: ArchLinux
 - Python version: 3.10.1
 - NanoVNA-Saver version: 0.3.10

RandMental, 2022-01-15:
This is fixed by the pull request I submitted a few days ago

Werzi2001, 2022-01-15:
@RandMental Thank you. Actually I tried adding int() as well after posting this report. It worked there but I just got other errors of similar kind. I think NanoVNA-Saver is just not compatible to Python 3.10. Or were you successful by just adding the int()?

RandMental, 2022-01-15:
That is interesting - (sweep) percentage is only used  at this point so it should not result in other errors by converting it to int before calling. setValueI am using pycharm and Python 3.10 and updated to 3.10.1.   Both worked fine and I tested the change with an H, H4 and V2plus4.   All three works fine with this change.   I am also running the latest version of the required libraries, and not necessarily the version required. Regards Sent from my Galaxy

zarath, 2022-01-15:
Numpy and scipy seems to be picky about lib versions. Also pyinstaller for generation of binaries is not yet available in version 4.8 on pypi - 4.7 doesn't support python 3.10. I have relaxed the version requirements in setup.cfg to allow easier installation on fedora. My builds an tests are done with the libs defined in requirements.txt.
Building with pyinstaller v4.8 from github and python 3.10 resulted in working binaries on my machine (Ubuntu 21.10)

Werzi2001, 2022-01-16:
Ok then I'll try using pyinstaller (which I haven't so far). Even with the int() fix I still get errors like
```
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 4655, resource id: 14687264, major code: 40 (TranslateCoords), minor code: 0
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Frequency.py", line 480, in paintEvent
    self.drawValues(qp)
  File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/RI.py", line 293, in drawValues
    qp.drawLine(x, y_re, new_x, new_y)
TypeError: arguments did not match any overloaded call:
  drawLine(self, QLineF): argument 1 has unexpected type 'int'
  drawLine(self, QLine): argument 1 has unexpected type 'int'
  drawLine(self, int, int, int, int): argument 3 has unexpected type 'numpy.float64'
  drawLine(self, QPoint, QPoint): argument 1 has unexpected type 'int'
  drawLine(self, Union[QPointF, QPoint], Union[QPointF, QPoint]): argument 1 has unexpected type 'int'
```

Maybe that's not related though.

RandMental, 2022-01-16:
Could be related. How did you do the Int()? Should be:self.sweep_control.progress_bar.setValue(int(self.worker.percentage))RegardsSent from my Galaxy

Werzi2001, 2022-01-16:
@RandMental Yes exactly like that. Anyhow the error is in a completely different line and therefore I guess there should be another int() being added but that seems to be very tedious. Maybe there is a different position to change the type of the value on creation and not on every single usage.

RandMental, 2022-01-16:
Agree.  

I am running  Pycharm 2021.3 latest update  with the following packages, not all are used in Saver:

 

APScheduler

3.8.1

3.8.1

Cython

0.29.26

0.29.26

Eel

0.12.4

0.14.0

MouseInfo

0.1.3

0.1.3

Pillow

8.4.0

9.0.0

PyAutoGUI

0.9.53

0.9.53

PyGetWindow

0.0.9

0.0.9

PyMsgBox

1.0.9

1.0.9

PyQt5

5.15.6

5.15.6

PyQt5-Qt5

5.15.2

5.15.2

PyQt5-sip

12.9.0

12.9.0

PyQt5-stubs

5.15.2.0

5.15.2.0

PyRect

0.1.4

0.1.4

PyScreeze

0.1.28

0.1.28

PyYAML

6.0

6.0

altgraph

0.17.2

0.17.2

appdirs

1.4.4

1.4.4

argparse

1.4.0

1.4.0

async-generator

1.10

	

attrs

21.2.0

21.4.0

auto-py-to-exe

2.13.0

2.15.0

beautifulsoup4

4.10.0

4.10.0

bottle

0.12.19

0.12.19

bottle-websocket

0.2.9

0.2.9

cachetools

4.2.2

5.0.0

certifi

2021.10.8

2021.10.8

cffi

1.15.0

1.15.0

charset-normalizer

2.0.9

2.0.10

consonance

0.1.2

0.1.5

cryptography

36.0.0

36.0.1

dissononce

0.34.3

0.34.3

dnspython

2.1.0

2.1.0

envelope

1.5.3

1.5.3

filelock

3.4.0

3.4.2

future

0.18.2

0.18.2

gevent

21.8.0

21.12.0

gevent-websocket

0.10.1

0.10.1

greenlet

1.1.2

1.1.2

h11

0.12.0

0.12.0

idna

3.3

3.3

iso8601

1.0.2

1.0.2

jsonpickle

2.0.0

2.1.0

keyboard

0.13.5

0.13.5

numpy

1.21.5

1.22.1

outcome

1.1.0

1.1.0

pefile

2021.9.3

2021.9.3

pip

21.3.1

21.3.1

protobuf

3.19.1

3.19.3

py3-validate-email

1.0.4

1.0.5

pyOpenSSL

21.0.0

21.0.0

pycparser

2.21

2.21

pyinstaller

5.0.dev0

4.8

pyinstaller-hooks-contrib

2021.4

2021.5

pyparsing

3.0.6

3.0.6

pyperclip

1.8.2

1.8.2

pyreadline

2.1

2.1

pyserial

3.5

3.5

python-axolotl

0.2.2

0.2.3

python-axolotl-curve25519

0.4.1.post2

0.4.1.post2

python-dateutil

2.8.2

2.8.2

python-gnupg

0.4.8

0.4.8

python-magic

0.4.24

0.4.24

python-magic-bin

0.4.14

0.4.14

python-telegram-bot

13.8.1

13.10

pytweening

1.0.4

1.0.4

pytz

2021.3

2021.3

pytz-deprecation-shim

0.1.0.post0

0.1.0.post0

pywhatkit

5.2

5.3

pywin32-ctypes

0.2.0

0.2.0

requests

2.26.0

2.27.1

scipy

1.7.3

1.7.3

selenium

4.1.0

4.1.0

serial

0.0.97

0.0.97

serial-button-manager

0.1.4

0.1.4

setuptools

59.5.0

60.5.0

six

1.10.0

1.16.0

sniffio

1.2.0

1.2.0

sortedcontainers

2.4.0

2.4.0

soupsieve

2.3.1

2.3.1

style

1.1.0

1.1.6

tornado

6.1

6.1

transitions

0.8.10

0.8.10

trio

0.19.0

0.19.0

trio-websocket

0.9.2

0.9.2

typecode-libmagic

5.39.210531

5.39.210531

tzdata

2021.5

2021.5

tzlocal

4.1

4.1

update

0.0.1

0.0.1

urllib3

1.26.7

1.26.8

wheel

0.37.0

0.37.1

whichcraft

0.6.1

0.6.1

wikipedia

1.4.0

1.4.0

wsproto

1.0.0

1.0.0

yowsup2

3.2.3

3.2.3

zope.event

4.5.0

4.5.0

zope.interface

5.4.0

5.4.0

ZakiuC, 2022-05-20:
I've run into this problem too

zarath, 2022-05-20:
> Ok then I'll try using pyinstaller (which I haven't so far). Even with the int() fix I still get errors like
> 
> ```
> qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 4655, resource id: 14687264, major code: 40 (TranslateCoords), minor code: 0
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/Frequency.py", line 480, in paintEvent
>     self.drawValues(qp)
>   File "/usr/lib/python3.10/site-packages/NanoVNASaver/Charts/RI.py", line 293, in drawValues
>     qp.drawLine(x, y_re, new_x, new_y)
> TypeError: arguments did not match any overloaded call:
>   drawLine(self, QLineF): argument 1 has unexpected type 'int'
>   drawLine(self, QLine): argument 1 has unexpected type 'int'
>   drawLine(self, int, int, int, int): argument 3 has unexpected type 'numpy.float64'
>   drawLine(self, QPoint, QPoint): argument 1 has unexpected type 'int'
>   drawLine(self, Union[QPointF, QPoint], Union[QPointF, QPoint]): argument 1 has unexpected type 'int'
> ```
> 
> Maybe that's not related though.
> 
> Edit: My versions (on ArchLinux) are numpy = 1.21.5-2 scipy = 1.7.3-3 which does not match the requirements.txt perfectly. Probably that's the issue?

Completely differnt codeline as the reported error.

zarath, 2022-05-20:
Original Error was fixed.
Comments below are at a totally different line of code.

closed 2022-05-20 by zarath