NanoVNA-Saver/nanovna-saver #467: Windows 11, VCP driver needed

kieths, 2022-01-26:
I tried several times to get a connection to my NanoVNA-F, with Windows 11 Pro x64.
Searching various sources, I got pointers to installing a driver, "CypressDriverInstaller_1.exe".  No dice.

In the end, what made the COM port show up under the NanoVNA-Saver, was installing the VCP driver.
https://ftdichip.com/drivers/vcp-drivers/

I now am able to open and run the NanoVNA-Saver.
I note that opening takes a long time, and that the USA ham band 1.25 M (220 MHz) is not in the selection list, but it's running.

PA3CCE, 2022-02-07:
After updating to Windows11, my NanoVNA-V2, SAA2 and LiteVNA did nomore connect, re-installing Cypress Serial port driver solved the problem !

rjordans, 2022-02-11:
Adding custom bands to the band overview is something you can do.  Go to the "Display Setup" (bottom left corner) and there you should be able to find the "Manage bands" button that lets you add/remove bands from the list.

closed 2022-04-01 by zarath