NanoVNA-Saver/nanovna-saver #473: Bluetooth API incompatible

PaulWDent, 2022-02-27:
After the start screen has blinked a minute or two, the error message comes up "Your machine is not compatible with BLuetooth API"
But were not using Bluetooth, are we?

Down load nanoVNA software version 3.8 to a Windows 8 64-bit machine, unzip, and hot the ,exe file.

The nanoVNA is only any use if it can be controlled by a PC because navigating the touch screen MMI is incomprehensible and unreliable

zarath, 2022-04-01:
v0.3.8 and bluetooth?
Stil happens with v0.4.0

Roger-Need, 2022-05-15:
This is a problem with recent versions of PyQT5 not working on Windows 8/8.1 .  Bluetooth is not needed in Saver so here is the fix...   You can solve it by going into PyQT5\Qt\bin and renaming QT5Bluetooth.dll to QT5Bluetooth.dll1

Then build as usual and it now works without errors on Win 8/8.1.   I have to do this with every new release of Saver.

brownbess76, 2022-05-15:
Similar issue as PaulWDent.
Do not have problem on v.0.3.9
Error: This Windows version (6.3.9600) does not support the required Bluetooth API.

Not sure how to find QT5Bluetooth.dll to rename it as suggested by Roger-Need